rust-osdev/uefi-rs #1: Set up CI

GabrielMajeri, 2017-11-22:
Steps:
- [ ] Find a way to pass an error code through QEMU.
- [ ] Make testing unattended.
- [ ] Make panics shutdown the VM.

HadrienG2, 2018-01-19:
If you want a quick and minimally invasive way to have QEMU communicate info to the CI host, I would suggest the port e9 hack, which is also supported by Bochs. On x86, it allows you to send bytes to the host just by writing to the (reserved and unused) e9 CPU port. You enable it by passing the `-debugcon` command line switch to QEMU.

GabrielMajeri, 2018-01-20:
@HadrienG2 I'll try that, and perhaps use Python to parse the test results.

I'll also see if it's possible to create a custom unit test harness so we can use the standard `#[test]` syntax.

FredrikAleksander, 2018-03-27:
There is actually a EFI protocol to interface with the debug port. It would be alot cleaner than I/O ports directly. When it comes to running the tests, this is really simple, just spawn qemu and specify a debug log file, then listen for writes on that log file, and whenever a message appears in the log that indicates the tests have completed, or a panic occured, kill the qemu process if it is still running, and parse the log. If panics always start with PANIC!: it is easy to determine when the VM has paniced. 

I do something similar after my builds, to determine where OVMF loads my image so I can get the addresses to load debug symbols at. OVMF outputs alot of debug info to the log, which i read until I get the results I need or a timeout occurs. I could help out with this if wanted

GabrielMajeri, 2018-03-27:
> There is actually a EFI protocol to interface with the debug port.
Hmm, wasn't aware of that. I'm a bit busy right now but I will look into it ASAP. Some unit tests run on every commit and pull request would definitely help.

GabrielMajeri, 2018-09-09:
I've discovered the [integration testing](https://os.phil-opp.com/integration-tests/#shutting-down-qemu) part of the "Writing an OS in Rust book", and it suggests we can use the `isa-debug-exit` port to write an error code and shut down the machine. We can simply check if the returned code is 0 or not.

Setting up on the command line would be:

```
-device isa-debug-exit,iobase=0xf4,iosize=0x04
```

HadrienG2, 2018-09-20:
Since OVMF spams the debug console, maybe an alternate strategy for test output would be to use a serial port. AFAIK, that's a common strategy for Linux debugging, and with EFI_SERIAL_IO_PROTOCOL available it might be relatively straightforward to implement.

HadrienG2, 2018-09-20:
Ah, wait, I see that it's spamming only because you attached it to a specific port for OVMF debugging. Makes sense. I also see that you removed support for the old style port E9 hack which you previously introduced.

Can you bring me up to speed with your current ideas for sending test output to the host? Maybe I can help?

GabrielMajeri, 2018-09-21:
@HadrienG2 
> removed support for the old style port E9 hack

I've disabled the old port E9 code because it wasn't used. It was simply there as a placeholder. I have no issue against it, except for the fact it might not work if we plan to also support other architectures UEFI supports (although I've only tested with x86 for now).

> EFI_SERIAL_IO_PROTOCOL

That might be interesting to try.

>  your current ideas for sending test output

I started work on the UEFI debug port protocol. It seems OVMF has support for sending stuff to GDB, but I realised that's very low-level and would probably require an implementation of GDB's remote debug protocol.

I also though that for the GOP-related tests you'd need a human to visually inspect the output anyway. So I though it wasn't worth the hassle and just stuck with simply outputting stuff to screen :)

I am also waiting to see where the [custom test frameworks RFC](https://github.com/rust-lang/rust/issues/50297) is going to see if we could eventually write tests with `#[test]` instead of manually running them.

GabrielMajeri, 2018-09-21:
@HadrienG2 Alright, then I guess the serial port it is. Let me know if you want to implement the `SERIAL_IO_PROTOCOL` yourself or if I can help.

I'll also see about how we can better organize the tests to possibly take advantage of the [custom test framework RFC](https://github.com/rust-lang/rust/issues/50297). Although at this early stage it didn't seem to work for me.

HadrienG2, 2018-09-21:
I think you'll be faster than me at interfacing the SERIAL_IO_PROTOCOL part. I only just got back into the UEFI spec, and am still mourning how huge it is. I thought the ACPI record was unbeatable, but Microsoft will never cease to surprise me...

GabrielMajeri, 2018-09-21:
@HadrienG2 Well it seems our work might be cut out for us. I've simply added `-serial stdio` to QEMU, without adding any serial device support, and look what happened:

[image]

[image]

QEMU automatically redirected UEFI's output to serial port / stdio! We just need to come up with a way to parse the text output with `grep` or something and it should work OK.

HadrienG2, 2018-09-21:
That's cool and scary at the same time :) As I guess it means that OVMF is silently sending our logs on the serial port under the hood. Not sure if that would be desirable if e.g. testing on an ARM development board. Anyhow, time to experiment some more !

HadrienG2, 2018-09-21:
In any case, I was not so lucky...

[image]

Now, I need to figure out why :)

GabrielMajeri, 2018-09-21:
@HadrienG2 I encountered that too, I guess it's a side-effect of enabling serial output. We should probably handle "unsupported feature" errors by just ignoring the test.

HadrienG2, 2018-09-21:
Agreed, except in cases where the availability of the functionality is part of the test (as in e.g. EFI_SIMPLE_TEXT_OUTPUT_PROTOCOL.TestString() ). I'll take a tour through the tests and adjust them accordingly.

HadrienG2, 2018-09-22:
So, I think my thoughts are starting to come together nicely and I have a reasonable plan for finishing the CI integration.

The first step is to put back the QEmu serial port in stdio mode. That is the most convenient way of communicating with the host which we've found so far: it's automatically fed with the output logs by OVMF, it's bidirectional (the host can send a reply back to the guest), it's easy to manipulate from the Python side, and with minor adaptations it can be adapted to tests on real hardware too. The only drawback of doing that is that unless we go down the path of using two serial ports, it breaks the recently introduced loopback serial tests. But as you will see, I have an idea for fixing that too.

Next step is to make every line of the logger's output start with a prefix which indicates what kind of log is being emitted. For example, every line of `info!()` logs could start with `"INFO: "` and every line of `error!()` logs could start with `"ERROR: "`. This has two positive effects. The first one is that we can easily discriminate what kind of log we're dealing with (and e.g. consider emission of an ERROR log as a test failure), which we couldn't do before. The second one is that now, by prefixing our serial output with something _other than a log level_ we have created a basic protocol for sending the host some commands/requests in addition to the logs.

The first command which we could introduce would be the `"ECHO: <text>"` command. The idea is simple: if the host sees an ECHO command, it sends the text after that command back the guest via QEmu's stdin. This allows us to bring back the serial loopback test in a pretty clean way.

The second command which we could introduce would be the `"CHECK_SCREENSHOT: <filename>"` command (command names are open to bikeshedding too). This command asks the host to take a QEmu screenshot (which can be done via the QEMU monitor), then compare it with a corresponding reference file in the test codebase. If the comparison succeeds, the host sends back an acknowledgement to the guest so that it can move on with other things, e.g. "OK". If the comparison fails, the host either kills QEMU or asks the guest to terminate cleanly with another reply (e.g. "KO").

Finally, at the end of a test run, the test application would send to the host a predefined message, e.g. "SUCCESS", which indicates that the test suite completed successfully. This means that we do not need to parse user-oriented output to reach such conclusions, which is always nice.

All these testing-oriented commands obviously rely on the fact that there is something listening at the other end of the UEFI standard output and serial port, so they should be compiled out with a cargo feature when testing on real hardware. But overall, I think that this design gives us a solid and versatile test harness that can easily be integrated into CI at a low implementation cost.

What do you think? Should I start implementing that?

GabrielMajeri, 2018-09-22:
@HadrienG2 Sounds like a great idea! I do believe we should start small and make sure at least the basic tests still work, excluding the GOP and the serial tests.

> unless we go down the path of using two serial ports

I don't see an issue with this, it's better to keep stuff separated.

 > the host sends back an acknowledgement to the guest

Or we can just let QEMU keep running and `kill` the process if we don't like the screenshot :)

> the test application would send to the host a predefined message

I think the `0` return code is enough. We already shut down the computer after a time-out at the end of the tests.

HadrienG2, 2018-09-22:
Ok, I'll start putting these ideas in practice then!

>> unless we go down the path of using two serial ports
>
> I don't see an issue with this, it's better to keep stuff separated.

One small drawback is that we then need to tell which serial port is which on the UEFI side, which may require keeping QEmu command flags in sync with the testing code. I would prefer keeping the QEmu configuration and the tests as independent possible.

Another drawback if we intend to extend the testing procedure to real hardware later on is that real hardware typically has only one physical serial port (even if _supporting_ multiple ones by e.g. having multiple UARTs in the chipset is common).

But these are minor concerns and I could quite easily be convinced to go with the "two ports" solution.

>> the host sends back an acknowledgement to the guest
>
> Or we can just let QEMU keep running and kill the process if we don't like the screenshot :)

Without acknowledgement, there is a risk of race condition though: if the host is not fast enough at taking the screenshot, the guest may have changed the display meanwhile. We can resolve the race in other ways (e.g. by sleeping), but I think host acknowledgment has the best cleanness/complexity ratio.

>> the test application would send to the host a predefined message
>
> I think the 0 return code is enough. We already shut down the computer after a time-out at the end of the tests.

It's true that as long as we're testing in QEMU, we don't need more than a status code.

GabrielMajeri, 2018-09-22:
@HadrienG2 Thanks for your contributions. Would you be interested in becoming a co-maintainer of this crate?

> Ok, I'll start putting these ideas in practice then!

You might want to wait until the [ci](https://github.com/GabrielMajeri/uefi-rs/tree/ci) branch gets done, I am working on getting QEMU set up in Travis.

commit fa0206fa (2018-09-22): Run tests with QEMU in Travis CI

This commit installs QEMU + OVMF in Travis, and uses
them to run the tests in CI.

The build script has been updated to allow running QEMU
in headless mode.

Fixes #1. Also fixes an issue with caching not working in CI.
closed 2018-09-22 by GabrielMajeri