rust-osdev/uefi-rs #3: exit_boot_services should consume the host BootServices

HadrienG2, 2018-02-24:
Since the boot services are not usable after calling exit_boot_services, they should not remain available in the API. So I would propose having BootServices::exit_boot_services take `self` as a parameter, not `&self`.

GabrielMajeri, 2018-02-26:
@HadrienG2 I currently have two ideas for solving this:

- Making the function `unsafe`, and have the user handle the unsafety. This is the simplest solution to implement, but requires the user to be really careful of what references they keep.

- Use a closure / callback which is only allowed to update the memory map, since that is the only function UEFI allows us to call after the first `exit_boot_services` call.

The issue is that it's not easy to "move" the boot services table into the function, since we only have a reference to it stored in the system table.

HadrienG2, 2018-02-26:
Given this extra info, maybe we could combine my original idea and your second idea as follows:

* The exit_boot_services method is moved to the SystemTable object in the Rust API (as that makes it more explicit that it also consumes portions of the system table).
* It takes a closure as a parameter, allowing the user to check the memory map.
* It consumes portions of the system table, through either of the following strategies:
    1. Turn these portions into Options, and set them to None
    2. Consume the SystemTable object and return a simpler RuntimeSystemTable object

One issue to be resolved is that in both of these cases, we need to somehow move or modify the SystemTable object, which we cannot do because UEFI only gives us a &'static SystemTable. Maybe we could approach this by having uefi-rs insert a small shim before the user entry point which builds a custom system table wrapper from the UEFI-provided pointer.

Or, as you say, we could also make exit_boot_services unsafe now, and leave that question for later.

toothbrush7777777, 2018-07-31:
But it is possible to re-enter boot services — as far as I remember — if you keep all the memory allocated by UEFI and reload the GDT with the UEFI one. Or something like that.

GabrielMajeri, 2018-07-31:
@toothbrush7777777 It is not possible.

Once `ExitBootServices` is called, there is no `StartBootServices` function or anything like that.

You might be thinking of *runtime services* - those services are available as long as the computer is running, and work as long as the OS informs UEFI of the updated memory map.

toothbrush7777777, 2018-07-31:
@GabrielMajeri Yes, perhaps I was thinking of that. Did they include services for shutting down / restarting the computer?

HadrienG2, 2018-09-21:
Looking at the spec again, I think I see a winner design emerging. Let's see what the spec has to say about ExitBootServices:

- The handle must be that of the running image.
- You should call GetMemoryMap immediately beforehand (to maximize odds of success).
- If the call fails, you are expected to call GetMemoryMap and try again (and calling any other boot service at this point in time is unsafe).
- After the call succeeds, the boot service table and a few fields of the system table become unavailable.

Given these constraints, it seems to me that the ideal interface would be a function which takes no parameters and...

- Takes the running image handle from wherever it's located (an entry point parameter, sadly)
- Repeatedly alternates between creating a memory map and calling exit_boot_services() until that succeeds
- Once that succeeds, prevent the user from accessing the old system and boot services tables (as

HadrienG2, 2018-09-21:
Sorry for the noise, Github's infamous "close and comment" button ergonomics bit me.

HadrienG2, 2018-09-23:
After reading through chapter 2 of the UEFI spec, and in particular pp 29-30, it looks like even runtime services have a pretty elaborate calling contract. So ideally we should figure out a way to make runtime services unsafe after ExitBootServices are called, to remind OSes of the need to uphold that contract.

I haven't given this much thought yet, but I think resolving the "consuming boot time services part" should directly allow us to provide a different interface to run time services too. It's the first bit that will take some thought :)

Another part which will take some thought is that exiting boot services invalidates most protocols in flight. I have a hunch that we might be able to leverage the Rust lifetime system to prevent dangling protocols here, but until I've tried, there's no way to tell if this actually works _and_ provides a practical interface.

HadrienG2, 2018-09-23:
I think a promising design would be to have a repr(transparent) newtype around an &'static SystemTable, and pretend that the UEFI entry point receives that as a parameter instead of the system table reference. repr(transparent) should ensure that this works, and the newtype should provide enough flexibility to implement "table-consuming semantics".

HadrienG2, 2018-10-14:
I've played around this a bit this week-end, and as a first step I've started clarifying the lifetimes in uefi-rs so that the borrow checker can analyze more (less meaningless `'static`s). I think I'm not that far away from a clean and safe interface for exiting boot services anymore, but only time will tell.

GabrielMajeri, 2018-10-15:
The `README` recommends people to only use UEFI for the boot loader part, and try to keep their real app as an ELF payload.
This way, the actual UEFI binary only runs until a kernel is loaded and `ExitBootServices` gets called, at which point only runtime service calls should get made.

UEFI is such a  "great" API people try to avoid using it as much as possible in practice.  And the less they interact with it, the lower the risk of errors / unsafety.

PR #65 (2018-10-23): Type-safety cleanup and safe exit from boot services
This PR gives uefi-rs a solid type-safety cleanup (e.g. no more "usize" as a substitute for pointers, and using 'static when that's a lie is avoided). Most importantly, however, it also fixes #3 by providing a safe method to exit UEFI boot services, ensuring that the borrow checker will warn you if you try to use boot services after the fact.

Achieving this result required a significant redesign of how protocols work. Previously, protocols were assumed to live forever, whereas now they are treated as a reference to UEFI boot services. Some  [... cut]
closed 2018-10-23 by HadrienG2