rust-osdev/uefi-rs #16: linker error on latest nightly

sdemos, 2018-07-12:
I just updated to the latest nightly today and I got this error - 

```bash
$ ./build.py build
   Compiling uefi-test-runner v0.1.0 (file:///home/demos/projects/uefi-rs/uefi-test-runner)
error: linking with `lld` failed: exit code: 1===================>      ] 9/10     
  |
  = note: "lld" "-flavor" "link" "/Subsystem:EFI_Application" "/Entry:uefi_start" "/LIBPATH:/home/demos/.xargo/lib/rustlib/x86_64-uefi/lib" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.114xigf1yyw2gcw0.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.16u6js6g0l3k1ic6.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.181cuta0v63atwcm.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.1c4sbqhvukbgthag.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.1dqvxks6k2bzkxe.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.1im38lueib99jsk0.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.1mvmz58owquyropc.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.1y16o1qfye96o7m0.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.23tqyymcb18u96mb.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.2fipfjjg6nxsyp4t.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.2imnh2vhxcqrizhm.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.2jqywn86b2gsqohu.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.2lyh15q6cjwzy18c.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.2mb8rq5da3exkx2g.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.2oiwwqohid6mabhc.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.2q5257pdh5222n7q.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.33j957i3px7pj5vz.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.35c7yhz90tqp9m9a.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.37huh28k6g3px6ck.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.38ps4pa181wsnsy9.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.391nabwpj8wu7mr0.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.3ayaeypdcro9d6yk.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.3s96syg39ip5dg5w.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.436dotimmrgzkwfa.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.43v6g0y2xsxoggnt.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.47chcpo31ood8q32.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.48721dc4k5qxei0u.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.49a7n47po4ttqjl7.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.4ahv1jxw4hkr9dtv.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.4ezmh1vbs95c5ack.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.4lmbw51bfoes4lkp.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.4xq48u46a1pwiqn7.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.4yh8x2b62dcih00t.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.4ypvbwho0bu5tnww.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.56dly8q07ws8ucdq.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.57k06xfugllsc526.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.5by9xggibmun5kn9.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.7p53qlves2rwxx0.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.8hvrv8qpny8dbel.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.8xzrsc1ux72v29j.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.9elsx31vb4it187.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.9fcb3syd3ne5k0n.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.bxp0med4do8g2u9.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.rzsapsfex8xntj4.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.y08g5q2x813c4wx.rcgu.o" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.z9ox7biyn1otfln.rcgu.o" "/OUT:/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.efi" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/uefi_test_runner-7bc0d1f33800b032.crate.allocator.rcgu.o" "/OPT:REF,NOICF" "/DEBUG" "/LIBPATH:/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps" "/LIBPATH:/home/demos/projects/uefi-rs/target/debug/deps" "/LIBPATH:/home/demos/.xargo/lib/rustlib/x86_64-uefi/lib" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/libuefi_utils-8050474b59705aeb.rlib" "/home/demos/.xargo/lib/rustlib/x86_64-uefi/lib/liballoc-fecacabd009468bc.rlib" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/libuefi_services-b91c7ff6cea21318.rlib" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/libuefi_alloc-0be210fde2ecf984.rlib" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/libuefi_logger-399a6cab2dc38960.rlib" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/liblog-d266a9b7833cdac5.rlib" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/libcfg_if-fe0d654391b4e273.rlib" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/librlibc-e741ceffc6c2dba7.rlib" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/libuefi-5ccfc5e76190e324.rlib" "/home/demos/projects/uefi-rs/target/x86_64-uefi/debug/deps/libbitflags-f75c9383c243bedd.rlib" "/home/demos/.xargo/lib/rustlib/x86_64-uefi/lib/libcore-d722c004db30ce6b.rlib" "/home/demos/.xargo/lib/rustlib/x86_64-uefi/lib/libcompiler_builtins-7fa11aff09ec40fa.rlib"
  = note: lld: error: libcore-d722c004db30ce6b.rlib(core-d722c004db30ce6b.core14-11a635b2ecf559281af1ed28854a3d65.rs.rcgu.o): undefined symbol: cosf
          lld: error: libcore-d722c004db30ce6b.rlib(core-d722c004db30ce6b.core14-11a635b2ecf559281af1ed28854a3d65.rs.rcgu.o): undefined symbol: cos
          lld: error: libcore-d722c004db30ce6b.rlib(core-d722c004db30ce6b.core14-11a635b2ecf559281af1ed28854a3d65.rs.rcgu.o): undefined symbol: sinf
          lld: error: libcore-d722c004db30ce6b.rlib(core-d722c004db30ce6b.core14-11a635b2ecf559281af1ed28854a3d65.rs.rcgu.o): undefined symbol: sin
          

error: aborting due to previous error

error: Could not compile `uefi-test-runner`.                                       

To learn more, run the command again with --verbose.
Subprocess xargo exited with error code 101
```

Interestingly, it also required me to actually install `lld` with my package manager, whereas before it just seemed to work (apparently without it?). I'm not sure what is causing this, but I'm having the same issue with my project as well now, since I use the same target spec.

sdemos, 2018-07-12:
well, the reason I had to install `lld` is because they apparently renamed the bundled `lld` to `rust-lld`. I added `"linker": "rust-lld",` to the target spec and now it uses the bundled one. 

unfortunately, that didn't fix the linking issue.

GabrielMajeri, 2018-07-12:
Fixed in the latest few commits, it should link now. Not sure either why it requires those trigonometric functions, I guess it's due to some upstream changes. They don't seem to be called, however.

closed 2018-07-12 by GabrielMajeri