rust-osdev/uefi-rs #18: Could not find `alloc` in `{{root}}`

HadrienG2, 2018-09-18:
Just tried giving uefi-rs a spin again after a hiatus from low-level dev, and I get these strange build errors on my first try at running the tests:

```
uefi-rs/uefi-test-runner> ./build.py run
   Compiling uefi-exts v0.1.0 (/home/hadrien/Bureau/Programmation/uefi-rs/uefi-exts)                                                                                                                                                                                                                                         
error[E0433]: failed to resolve. Could not find `alloc` in `{{root}}`                                                                                                                                                                                                                                                        
 --> uefi-exts/src/boot.rs:5:5                                                                                                                                                                                                                                                                                               
  |                                                                                                                                                                                                                                                                                                                          
5 | use alloc::vec::Vec;                                                                                                                                                                                                                                                                                                     
  |     ^^^^^ Could not find `alloc` in `{{root}}`                                                                                                                                                                                                                                                                           
                                                                                                                                                                                                                                                                                                                             
error[E0433]: failed to resolve. Use of undeclared type or module `Vec`                                                                                                                                                                                                                                                      
  --> uefi-exts/src/boot.rs:26:26                                                                                                                                                                                                                                                                                            
   |                                                                                                                                                                                                                                                                                                                         
26 |         let mut buffer = Vec::with_capacity(buffer_size);                                                                                                                                                                                                                                                               
   |                          ^^^ Use of undeclared type or module `Vec`                                                                                                                                                                                                                                                     
                                                                                                                                                                                                                                                                                                                             
error[E0412]: cannot find type `Vec` in this scope                                                                                                                                                                                                                                                                           
  --> uefi-exts/src/boot.rs:11:51                                                                                                                                                                                                                                                                                            
   |                                                                                                                                                                                                                                                                                                                         
11 |     fn find_handles<P: Protocol>(&self) -> Result<Vec<Handle>>;                                                                                                                                                                                                                                                         
   |                                                   ^^^ not found in this scope                                                                                                                                                                                                                                           
                                                                                                                                                                                                                                                                                                                             
error[E0412]: cannot find type `Vec` in this scope                                                                                                                                                                                                                                                                           
  --> uefi-exts/src/boot.rs:18:51                                                                                                                                                                                                                                                                                            
   |                                                                                                                                                                                                                                                                                                                         
18 |     fn find_handles<P: Protocol>(&self) -> Result<Vec<Handle>> {                                                                                                                                                                                                                                                        
   |                                                   ^^^ not found in this scope                                                                                                                                                                                                                                           
                                                                                                                                                                                                                                                                                                                             
warning: unused import: `alloc::vec::Vec`                                                                                                                                                                                                                                                                                    
[... cut ...]
error: Could not compile `uefi-exts`.                                                                                                                                                                                                                                                                                        

To learn more, run the command again with --verbose.
Subprocess cargo exited with error code 101
```

I guess that for some reason, my toolchain does not find the standard alloc crate. Ever got this one and know how to handle it?

HadrienG2, 2018-09-18:
Possibly related things that I found around : https://github.com/rust-lang/rust/issues/54006 and https://github.com/rust-lang/rust/pull/54116

mopp, 2018-09-19:
Hi
I have faced this issue also.
Here is my snippet to reproduce this errors

```rust
// lib.rs
#![feature(alloc)]
#![feature(allocator_api)]
#[no_std]

extern crate alloc;

use alloc::alloc::{Alloc, AllocErr, Layout};
```

it causes the error for `edition = "2018"`.
it can be compiled with unused warnings when `edition = "2018"` is commented out.

mopp, 2018-09-19:
Sorry, I incorrectly commented. (I deleted it)

GabrielMajeri, 2018-09-19:
@HadrienG2 Thanks, it was indeed rust-lang/rust#54116.

Fixed on `master` in 994bfd620acd.

closed 2018-09-19 by GabrielMajeri