rust-osdev/uefi-rs #20: Panic hook seems to be optimized out in release builds

HadrienG2, 2018-09-20:
On my system, building the test runners with the --release flag results in panics being ignored, most likely because the panic handler is optimized out. This is probably not what we want, especially as returning from a `!` function is undefined behavior in Rust.

GabrielMajeri, 2018-09-23:
Commit 25765569bb6af9ba044364923e5ca308bc022262 added a [`hlt`](https://github.com/GabrielMajeri/uefi-rs/blob/25765569bb6af9ba044364923e5ca308bc022262/uefi-services/src/lib.rs#L147) in the `panic_handler`'s loop, using `asm!("volatile")`, and it seems the function will not be optimized out anymore. Could you check if this is still an issue?

HadrienG2, 2018-09-23:
Yup, it works for me now :+1:

closed 2018-09-23 by HadrienG2