rust-osdev/uefi-rs #41: Investigate efiapi!{} macro

HadrienG2, 2018-09-24:
In Rust, I think resolving #39 entails generating different extern "<something>" statements depending on the target architecture. The pointer wrapping discussed in #40, if it ends up being approved and implemented, could also use a bit of code generation to automatically turn *const T into EfiPtr<T> and *mut T into EfiPtrMut<T>.

A general solution to handle this kind of EFI API boilerplate would be to borrow the EFIAPI design from the UEFI spec and turn it into a Rust macro. For example, this code...

```rust
efiapi! { fn(u64, usize) -> Status }
```

...would generate this code on x86_64:

```rust
extern "win64" fn(u64, usize) -> Status
```

There are limitations to what a declarative macro can generate in Rust, and we probably don't want to go for procedural macros on a task this simple, so the final design may be modified to accomodate these limitations. But I think the general idea is a sound one, which could address the issue of EFI calling conventions without bringing too much boilerplate in the declaration of each individual API entry point.

GabrielMajeri, 2018-09-24:
> extern "stdcall" fn(u64, usize) -> Status

But that's the thing, this is supposed to be `extern "C"` on i386, and everywhere else actually. UEFI spec wanted to have all functions just use `extern "C"` on every platform, but x86_64 is the only place where we have this weird separation of Windows vs UNIX ABI

> borrow the EFIAPI design

Using a macro seems to sort of go around the real problem. I would rather raise an issue upstream about this.

In [this old-ish comment](https://github.com/rust-lang/rust/issues/11261#issuecomment-52151873) it is mentioned that the only reason Rust even exposes the `win64` ABI is _because UEFI requires it_.

I'm sure the Rust devs would be open to a proposal to add an `extern "uefi"` or something ABI, if we can justify that it is useful.

HadrienG2, 2018-09-24:
> I'm sure the Rust devs would be open to a proposal to add an `extern "uefi"` or something ABI, if we can justify that it is useful.

Sounds like a good idea! I opened an issue on rust-lang to test waters, let's see how this goes...

HadrienG2, 2018-10-14:
After further investigation, this is too much for declarative macros (as they cannot expand into types, but only into items, the whole struct definition would need to be parsed, which is too much work) and too little to justify a procedural macro, so I'll live with the code duplication for now :wink:

closed 2018-10-14 by HadrienG2