rust-osdev/uefi-rs #45: I do not understand BltPixel's definition

HadrienG2, 2018-09-25:
The current spec has the following to say about the Blt buffer's pixel format (pp. 489)

> Each software Blt buffer entry represents a pixel that is comprised of a 32-bit quantity. Byte zero of
the Blt buffer entry represents the Red component of the pixel. Byte one of the Blt buffer entry
represents the Green component of the pixel. Byte two of the Blt buffer entry represents the Blue
component of the pixel. Byte three of the Blt buffer entry is reserved and must be zero.

I would expect that definition to translate into this struct:

```rust
#[repr(C, packed)]
pub struct BltPixel {
    pub red: u8,   // "Byte zero represents the Red component"
    pub green: u8,  // "Byte one represents the Green component"
    pub blue: u8,   // "Byte two represents the Blue component"
    _reserved: u8,  // "Byte three is reserved and must be zero"
}
```

Or maybe this one, if I badly misunderstood UEFI's endianness, into this:

```rust
#[repr(C, packed)]
pub struct BltPixel {
    _reserved: u8,  // "Byte three is reserved and must be zero"
    pub blue: u8,   // "Byte two represents the Blue component"
    pub green: u8,  // "Byte one represents the Green component"
    pub red: u8,   // "Byte zero represents the Red component"
}
```

However, what uefi-rs does is this:

```rust
#[repr(C, packed)]
pub struct BltPixel {
    pub blue: u8,   // "Byte two represents the Blue component"
    pub green: u8,  // "Byte one represents the Green component"
    pub red: u8,   // "Byte zero represents the Red component"
    _reserved: u8,  // "Byte three is reserved and must be zero"
}
```

There is no question that it works, and that the other interpretations don't (just gave them a try). But I don't understand why. It feels like there is a mismatch between what the spec says and what the implementation (here OVMF) is doing. Can you enlighten me?

GabrielMajeri, 2018-09-26:
I haven't even read that part of the spec.

I just scrolled down a few pages to the `EFI_GRAPHICS_OUTPUT_PROTOCOL.Blt()` section and found this:

```c
typedef struct {
UINT8 Blue;
UINT8 Green;
UINT8 Red;
UINT8 Reserved;
} EFI_GRAPHICS_OUTPUT_BLT_PIXEL;
```

I'm a simple person, I see a structure definition, I copy it :)

<hr/>

But in all seriousness, BGRA is what modern GPUs use internally, due to the legacy of Windows' GDI, so it wouldn't even make sense to expose a buffer in RGBA or ABGR order. 

I'm guessing either the UEFI spec got the wording wrong, or when they say "byte 0" they mean "the byte indicated by the first element of the `EFI_PIXEL_BITMASK` structure".

HadrienG2, 2018-09-26:
Yeah, my first a spec bug! :smile: Reported to the UEFI forum, thanks for the explanation.

closed 2018-09-26 by HadrienG2