rust-osdev/uefi-rs #54: FileMode and FileAttribute not exported

gz, 2018-09-27:
Even tough FileMode and FileAttribute are public:
* https://github.com/GabrielMajeri/uefi-rs/blob/9bce39c5dbb6299d4ead161c06ae4b1721a5ceec/src/proto/media/file.rs#L176
* https://github.com/GabrielMajeri/uefi-rs/blob/9bce39c5dbb6299d4ead161c06ae4b1721a5ceec/src/proto/media/file.rs#L184

they are not exported to external users of the library:
* https://github.com/GabrielMajeri/uefi-rs/blob/9bce39c5dbb6299d4ead161c06ae4b1721a5ceec/src/proto/media/mod.rs#L7

Is there another way to get access to these bitfields that I'm missing?

HadrienG2, 2018-09-28:
I think this is just a visibility oversight, which wasn't seen because this part doesn't have tests yet. We should probably fix both of these problems at the same time.

To prevent filesystem tests from having a long-lasting impact on the testing environment (even if there's a bug in them), it would be good to have them operate on a throwaway ramdisk which QEMU discards at the end.

HadrienG2, 2018-09-28:
@GabrielMajeri I have a number of things enqueued at the moment and will not be very available in the coming days, feel like taking this one?

PR #56 (2018-09-28): Make file-related structures and enums public
Also refactors these modules and expands documentation a bit.

Fixes #54.
closed 2018-09-28 by GabrielMajeri