rust-osdev/uefi-rs #66: Propose for inclusion in the rust-osdev org?

HadrienG2, 2018-12-05:
I just remembered that phil-opp is trying to build a [rust-osdev](https://github.com/rust-osdev) GitHub organization. Do you think that uefi-rs should join it for the benefit of some extra visibility ?

GabrielMajeri, 2018-12-05:
It would be awesome to get on board with the other Rust os-devs. If they're open to uefi-rs joining them, I see nothing bad with this. Who do we contact?

HadrienG2, 2018-12-05:
The rust-osdev gitter channel, according to the org's about page.

phil-opp, 2018-12-05:
Great to hear that! I would be more than happy if you decide to join rust-osdev!

You would of course keep full admin rights for the repo and would be able to move it out of the org again if you like. We could also create an UEFI team in the organization so that you have your own team-internal discussion page. Let me know if you have any questions!

phil-opp, 2018-12-06:
Awesome!

> I'm not entirely sure how to add the repo to it.

I added it for you. I'm not sure what's the required permission level, maybe only organization owners can do this.

I also gave you back the admin permissions for uefi-rs. Seems like GitHub set your access level only to "Write" after the move. Now you should have full control of the repo like before.

GabrielMajeri, 2018-12-06:
Ok, great!

closed 2018-12-06 by GabrielMajeri