rust-osdev/uefi-rs #67: Release on crates.io?

phil-opp, 2018-12-06:
Is there a reason why this crate is not released on crates.io?

GabrielMajeri, 2018-12-06:
On one hand, we're still doing some fundamental modifications to the code, so the interface is far from stable. I know that `0.x` versions don't guarantee any sort of stability, but we would be releasing new versions way too often. There would've been no advantage to labeling releases with semver.

Another issue is the name of the package. A [lot](https://crates.io/crates/uefi) [of](https://crates.io/crates/efi) the [good names](https://crates.io/crates/libefi) have been already taken.

phil-opp, 2018-12-06:
Thanks for the clarification!

About the name: The `uefi` crate wasn't updated since mid 2017, maybe the author is willing to transfer the name.

lweb20, 2019-04-13:
+1 for publishing unstable versions

edigaryev, 2019-09-21:
>On one hand, we're still doing some fundamental modifications to the code, so the interface is far from stable.

https://github.com/rust-osdev/uefi-rs/graphs/code-frequency

Is this still the case after one year of development?

Also, releasing this crate will automatically make documentation available on [docs.rs](http://docs.rs/) which is really handy for development, as currently you have to build it manually on each development machine.

GabrielMajeri, 2019-09-21:
@edigaryev I agree that the core API is more-or-less stable and I'd love to have easily accessible docs too.

But these are just excuses, it's more my laziness that's to blame :/

I've contacted the person who owns the `uefi` name on crates.io, and hopefully they will give us the name. Otherwise, I will start thinking about new names for the crate and work towards publishing it.

GabrielMajeri, 2019-09-22:
Great news! The author of the other crate replied and provided us the name. We're now published on [crates.io](https://crates.io/crates/uefi)! And the [docs.rs page](https://docs.rs/uefi/) works too!

As part of this, I've eliminated the `uefi_alloc` and `uefi_logger` crate and instead made them into optional features of the core crate. This will simplify publishing / dependencies and I see no potential downsides. Instead of `uefi_alloc::` the users now have to use `uefi::alloc::` (and enable the option in `Cargo.toml`). It gives me the idea to keep on simplifying and possibly remove the `exts` and `services` crates too.

closed 2019-09-22 by GabrielMajeri