rust-osdev/uefi-rs #70: Maybe uefi::Result should allow for custom errors

HadrienG2, 2018-12-12:
Providing extra data about an error have been needed a couple of times now. Usually, the error data is only available for specific error types. It would be nice if we didn't need to fallback to `core::result::Result` whenever that is needed, because in doing so we lose ergonomics conveniences, treat warnings as errors, and confuse users by giving them two different Result types.

Here are some rough ideas:

```rust
// Directly maps the semantics, but implementing a custom error requires work
type Result<T, E: From<Status>> = core::result::Result<Completion<T>, E>;

// We could ease this work by providing something like this
impl<ExtraErrData> From<Status> for (Status, Option<ExtraErrData>) { ... }

// Or, if we feel the Option will always be good enough, flatten the abstraction
type Result<T, ExtraErrData> = core::result::Result<Completion<T>, (Status, Option<ExtraErrData>)>;

// The question is, do we sometimes want this, which is closest to today's practice ?
type Result<T, ExtraErrData: Default> = core::result::Result<Completion<T>, (Status, ExtraErrData)>;

GabrielMajeri, 2018-12-13:
I think at this point we're getting into the more general, unresolved Rust issue of how to design composable error types.

The community already went through the standard `Error` type, error-chain, failure-rs, etc. and I still don't know which solution works best.

HadrienG2, 2018-12-13:
From what I have seen so far, every popular attempt at a general-purpose error handling abstraction uses dynamic memory allocation for type erasure. I don't think this is a luxury that we can afford at the core library layer (though we might integrate it in a higher-level layer like uefi-services), so we may need an abstraction that is more specialized towards UEFI's error-handling needs.

PR #78 (2019-01-04): Additional error data in Result + some API cleanup and clarification
This PR fixes #70 by adding support for additional error data to our Result type. As a result, usage of raw `core::result::Result` can now be restricted to functions which do not call into UEFI.

Since most functions do not return additional error data, the associated type parameter is optional with a default of `()`. After some thought, I think that the main output should probably get the same treatment, as a large amount of "setter" UEFI functions do not emit anything more than a status code.

While I went around the codeba [... cut]
closed 2019-01-04 by HadrienG2