rust-osdev/uefi-rs #72: Memory map queries do not enforce correct alignment

HadrienG2, 2018-12-13:
Both rustc and UEFI expect pointers to be correctly aligned for the target type. However, we do not enforce this when querying a memory map, as we take an unrestricted `&mut [u8]`, cast it to `*mut MemoryDescriptor`, and pass it down to UEFI. This is undefined behavior waiting to happen.

Since type alignments are small and bytes are cheap in the archs targeted by UEFI, the right thing to do in this case is probably to skip the first few bytes of the input slice in order to reach a part which is properly aligned, and do the write there.

I solved a very similar problem as part of #69. As I expect such "take a buffer as input" patterns to be common in UEFI's low-level environment, it might be worthwhile to take the alignment correction code that I wrote there and generalize it.

Here is a tentative API mockup:

```rust
/// Best API most of the time
fn realign_for<T: Sized>(storage: &mut [u8]) -> &mut [u8];

/// Lower-level API that is sometimes needed because mem::align_of does not support DSTs
fn realign(storage: &mut [u8], alignment: usize) -> &mut [u8];
```

Better DST support could be provided with a custom derive, that computes the alignment automatically instead of having people compute it by hand. But I would only do this at a later point in uefi-rs' evolution, if we encounter the use case sufficiently often to justify the significant code complexity of that infrastructure.

GabrielMajeri, 2018-12-13:
I would rather have a development-time mechanism which warns the developer that they somehow passed an unaligned buffer, yes.

The realignment functions could be useful as utilities for the users of the crate, but it feels like doing it silently is not beneficial.

HadrienG2, 2018-12-13:
Just to be clear, this calls for more attention than a development-time warning. Passing an unaligned buffer to UEFI or using an unaligned reference to Rust is undefined behavior and like any UB it is _more_ dangerous in release mode than in debug mode. So even in release mode, we must either fix it ourselves, panic when we encounter it, or mark functions that may take unaligned buffers as unsafe.

HadrienG2, 2018-12-14:
Shall we settle on documenting that the input slice must be well-aligned and panicking at runtime if it isn't?

GabrielMajeri, 2018-12-15:
It does seem reasonable to me. It should be the responsibility of the caller to handle memory allocation and alignment.

PR #76 (2018-12-21): Require correct alignment, but do not enforce it
Fixes #72 .
closed 2018-12-21 by HadrienG2