rust-osdev/uefi-rs #107: RuntimeServices SetVirtualAddressMap

gz, 2019-11-11:
I didn't find a way to call SetVirtualAddressMap through the runtime services in uefi-rs. Any reason why it is not exposed?

HadrienG2, 2019-11-11:
As often with missing functions in uefi-rs, the answer is most likely that it isn't interfaced yet. UEFI is a huge API, and interfacing all of it would be a matchingly large undertaking which none of us has enough time to dedicate to. Therefore, the current API exposed by this crate represents whatever the crate's maintainers found useful so far, and `SetVirtualAddressMap` [quite obviously didn't make that cut yet](https://github.com/rust-osdev/uefi-rs/blob/master/src/table/runtime.rs#L19).

If you find it useful for your work, patches interfacing it into `uefi-rs` are much welcome!

gz, 2019-11-11:
Thanks for clarifying, if it's just a missing wrapper I'll look into it and submit a patch.

PR #115 (2020-01-13): Add `set_virtual_address_map` interface in runtime services
According to the UEFI spec, `SetVirtualAddressMap` is used by the OS loader to map the runtime service memory region to virtual addresses after `ExitBootServices` is called. A bootloader will normally map the kernel address space to the higher-half and then jump to the kernel, and we can't leave the `RUNTIME_SERVICES_CODE` and `RUNTIME_SERVICES_DATA` alone. However, this process includes some absolute address relocation done by the runtime services themselves, which is an event triggered by the `SetVirtualAddressMap` call. Thus, `SetV [... cut]
closed 2020-01-13 by GabrielMajeri