rust-osdev/uefi-rs #133: Link fails on latest cargo-xbuild due to duplicate symbols (memset, memcpy etc.)

imtsuki, 2020-05-13:
Recent versions (0.5.30 and above) of `cargo-xbuild` enabled `-Clinker-plugin-lto` for the sysroot build. However `rust-lld` complains about duplicate symbols in `rlibc` and `compiler_builtins`, which are `memcpy`, `memmove`, `memset` and `memcmp`:

```
  = note: rust-lld: error: duplicate symbol: memcpy in librlibc-dfa6295423d90fe3.rlib(rlibc-dfa6295423d90fe3.rlibc.by8yst2e-cgu.0.rcgu.o) and in libcompiler_builtins-50c932690e366f4e.rlib(compiler_builtins-50c932690e366f4e.compiler_builtins.2wvp5uak-cgu.79.rcgu.o)
          rust-lld: error: duplicate symbol: memmove in librlibc-dfa6295423d90fe3.rlib(rlibc-dfa6295423d90fe3.rlibc.by8yst2e-cgu.0.rcgu.o) and in libcompiler_builtins-50c932690e366f4e.rlib(compiler_builtins-50c932690e366f4e.compiler_builtins.2wvp5uak-cgu.79.rcgu.o)
          rust-lld: error: duplicate symbol: memset in librlibc-dfa6295423d90fe3.rlib(rlibc-dfa6295423d90fe3.rlibc.by8yst2e-cgu.0.rcgu.o) and in libcompiler_builtins-50c932690e366f4e.rlib(compiler_builtins-50c932690e366f4e.compiler_builtins.2wvp5uak-cgu.79.rcgu.o)
          rust-lld: error: duplicate symbol: memcmp in librlibc-dfa6295423d90fe3.rlib(rlibc-dfa6295423d90fe3.rlibc.by8yst2e-cgu.0.rcgu.o) and in libcompiler_builtins-50c932690e366f4e.rlib(compiler_builtins-50c932690e366f4e.compiler_builtins.2wvp5uak-cgu.79.rcgu.o)
```

I found that there is an `extern` reference to `rlibc` in `uefi-services`:

https://github.com/rust-osdev/uefi-rs/blob/5814c52e6858ee72afd21339ca75d7297c474be9/uefi-services/src/lib.rs#L18-L19

But as [the homepage](https://github.com/alexcrichton/rlibc) of `rlibc` says, `rlibc` is now deprecated in favor of `compiler_builtins`, which provides the same functions. Is this a relic that we forgot to remove?

GabrielMajeri, 2020-05-13:
Thank you for your detailed analysis! Yes, that `rlibc` reference is from the time the crate was still required for the `mem*` functions. I will publish a fix shortly.

closed 2020-05-13 by GabrielMajeri

phil-opp, 2020-05-14:
Alternatively, it is possible to keep `rlibc` and use cargo's own `-Zbuild-std` flag. The easiest way to do this is to define aliases in a `.cargo/config` files, e.g. `xbuild = build -Zbuild-std=core,alloc`.