rust-osdev/uefi-rs #136: Verification of protocol addition confusion

ethindp, 2020-06-24:
So, I was looking at the MPServices protocol to see how it was done. I've got the UEFI spec open (v. 2.8). Assume I want to add support for the EFI DHCPv4 Protocol (section 29.2). This has a GUID of 8a219718-4ef5-4761-91c8c0f04bda9e56 (I think I wrote it right):
```C
#define EFI_DHCP4_PROTOCOL_GUID \
{0x8a219718,0x4ef5,0x4761,\
{0x91,0xc8,0xc0,0xf0,0x4b,0xda,0x9e,0x56}}
```
And a structure that looks like:
```C
typedef struct _EFI_DHCP4_PROTOCOL {
EFI_DHCP4_GET_MODE_DATA GetModeData;
EFI_DHCP4_CONFIGURE Configure;
EFI_DHCP4_START Start;
EFI_DHCP4_RENEW_REBIND RenewRebind;
EFI_DHCP4_RELEASE Release;
EFI_DHCP4_STOP Stop;
EFI_DHCP4_BUILD Build;
EFI_DHCP4_TRANSMIT_RECEIVE TransmitReceive;
EFI_DHCP4_PARSE Parse;
} EFI_DHCP4_PROTOCOL;
```
With each function defined as outlined in section 29.2 and each related structure also defined therein. What is the proper/correct (but not necessarily safe) way of making this callable from Rust?

HadrienG2, 2020-06-24:
Protocol declarations look like this:

```
#[repr(C)]
#[unsafe_guid("3fdda605-a76e-4f46-ad29-12f4531b3d08")]
#[derive(Protocol)]
pub struct SomeEfiProtocol {
    some_entry_point: extern "efiapi" fn(
        this: *const SomeEfiProtocol,
        some_param: SomeType,
        some_other_param: SomeOtherType,
    ) -> SomeResult,
    some_other_entry_point: extern "efiapi" fn(
        this: *const SomeEfiProtocol,
        yet_another_param: YetAnotherType,
    ) -> SomeOtherResult,
}
```

The general procedure for interfacing a protocol is as follows:

- Create some room in the uefi-rs folder hierarchy (which follows the UEFI spec hierarchy) for your protocol.
- Find out the signature of your entry points (i.e. what `EFI_DHCP4_START` maps to).
- If they contain some unknown types, build their `repr(C)` equivalent. Keep in mind that C's enums are different from Rust's.
- Build a struct with all the attributes of the examples above, that represents your protocol. The GUID goes in the `#[unsafe_guid(...)]` attribute.
- Fill in the signatures of your entry points as `extern "efiapi"` function pointers.
- Add an impl block for that struct with (ideally safe) public Rust functions that invoke those function pointers.

ethindp, 2020-06-24:
Thanks! It would be nice if this process was documented; perhaps an
example of an existing (simple) protocol like the console protocol,
maybe?

HadrienG2, 2020-06-24:
That's true. On the other hand, if I just copied the summary above, it might also give a false sense of simplicity, where there can actually be quite a bit of protocol-specific thinking involved to correctly interface a given struct or protocol.

Writing a more in-depth guide which warns about the various caveats to keep in mind (module organization, C enums, panics, safe interfaces that aren't, forgotten repr(C), aliasing rules...) is definitely above my time budget though. I wish I knew of an FFI guide with good coverage of those issues that I could just link to.

ethindp, 2020-06-24:
True. I wish I could do it, but I'm definitely going to be learning
these FFI interfaces as I go (and probably breaking things too... good
thing we have Qemu).

GabrielMajeri, 2020-06-24:
I've updated the [CONTRIBUTING.md](https://github.com/rust-osdev/uefi-rs/blob/master/CONTRIBUTING.md) file with the steps outlined by @HadrienG2 

@ethindp please tell us if there's anything else that should be documented :+1:

GabrielMajeri, 2020-07-02:
I'm going to close this for now; feel free to reopen.

closed 2020-07-02 by GabrielMajeri