rust-osdev/uefi-rs #138: device_handle of LoadedImage is private

toku-sa-n, 2020-07-14:
AFAIK getting SimpleFileSystem protocol needs device_handler of LoadedImage, but it's not marked as public. Is there another way to get SFS protocol?

```rust
fn prepare_file_system(image: &Handle, system_table: &SystemTable<Boot>) -> file::Directory {
    let loaded_image = system_table
        .boot_services()
        .handle_protocol::<loaded_image::LoadedImage>(*image)
        .expect_success("Failed to load image");

    let simple_file_system = system_table
        .boot_services()
        .handle_protocol::<fs::SimpleFileSystem>(loaded_image.into_inner().device_handle) // error: field `device_handle` of struct `uefi::proto::loaded_image::LoadedImage` is private
        .expect_success("Failed to prepare simple file system.");

    let simple_file_system = unsafe { &mut (*simple_file_system.get()) };

    simple_file_system
        .open_volume()
        .expect_success("Failed to open volume.")
}
```

HadrienG2, 2020-07-14:
None that I know of. AFAIK, it is safe to read device handles, but not to write to them, so you may want to resolve this by simply adding a read-only accessor to `LoadedImage` that returns the device handle.

GabrielMajeri, 2020-07-14:
I've added a getter for the respective handle in 522b7287addc52e3aba940ccf0a5ab5dd3b79955. Will also publish a new version of the crate with this change.

closed 2020-07-14 by GabrielMajeri