rust-osdev/uefi-rs #143: Rutime Services are available both before and after ExitBootServices

TrueBers, 2020-07-18:
https://github.com/rust-osdev/uefi-rs/blob/0c3999e6ee9b7d8a69edb19bde73c798b696c799/src/table/system.rs#L189-L191

There is a comment above stated that Runtime Services are available *only after* exiting the Boot Services.
But `UEFI Spec 2.8B May 2020` in chapter 8 states:

> **Runtime Services**. Functions that are available *before and after* any call to `ExitBootServices()`.

To clarify: the comment in code says that it's available **only after** exiting the Boot Services. The Specs say about both **before and after** cases.

I'm just started learning both Rust and UEFI, so I'm sorry if I'm wrong, but any chance my point is correct in this case or am I missing something?

GabrielMajeri, 2020-07-18:
The runtime services can indeed be used both before and after calling `exit_boot_services`.

However, this distinction between `SystemTable<Boot>` and `SystemTable<Runtime>` is not part of the UEFI API, it's just a type-level trick we use to maintain safety.

We needed a way to distinguish between the pre-exit-boot-services environment (where you can call both the boot and runtime services) and the post-exit-boot-services-environment (where you can only call runtime services).

The [SystemTable docs](https://docs.rs/uefi/0.4.7/uefi/table/struct.SystemTable.html) provides some more details on this transition.

HadrienG2, 2020-07-19:
From what I can see, this is just a docs wording issue. I will clarify it.

PR #145 (2020-07-19): Clarify SystemTable docs
Fixes #143
commit 4c2db945 (2020-07-19): Clarify docs, fixes #143
closed 2020-07-19 by GabrielMajeri