rust-osdev/uefi-rs #144: uefi-rs APIs which want a well-aligned buffer but accept &mut [u8] are unsafe

HadrienG2, 2020-07-19:
If the user-provided slice doesn't actually match the expected alignment, undefined behavior would ensue.

FWIW, many of these buffer-wranging APIs actually want to receive an *mut [u8], because they manipulate uninitialized memory and that concept doesn't exist in Safe Rust.

GabrielMajeri, 2020-07-19:
Would having some assertions for each such respective function ensure safety?

HadrienG2, 2020-07-19:
It would ensure safety in the presence of misaligned buffers (since [u8] itself has no such requirements), but using uninitialized memory as returned by e.g. UEFI's memory allocator would still be UB (although I don't think rustc can harmfully exploit this particular UB at the moment).

kotauskas, 2021-03-02:
The unitialized memory UB can be solved by using `&mut [MaybeUninit<u8>]` You can go a bit further and just use `&mut [MaybeUninit<MemoryDescriptor>]` in [`memory_map()`].

[`memory_map()`]: https://docs.rs/uefi/0.8.0/uefi/table/boot/struct.BootServices.html#method.memory_map " "

GabrielMajeri, 2021-03-02:
@kotauskas the idea with `&mut [MaybeUninit<u8>]` could work. The one with `&mut [MaybeUninit<MemoryDescriptor>]` wouldn't necessarily, since the `memory_map` API doesn't guarantee the returned memory descriptors are contiguous / occupy a fixed amount of spaced (you need to check each one to get the offset of the next.)

Soveu, 2021-03-03:
Currently `memory_map` would panic if the buffer is not aligned, but it could just take a slice that is aligned
```rust
let buf = &mut buf[buf.as_ptr().align_offset(8)..];
```

hanmertens, 2021-03-18:
Alternatively, a new type could be defined to represent a possibly uninitialized buffer with certain alignment requirements, e.g.:

```rust
#[repr(C, align(8))]
pub struct MemoryMapBuffer([MaybeUninit<u8>]);
```

A `&mut MemoryMapBuffer` could then be constructed unsafely from a pointer and length, or safely from a byte slice. In the former case the caller is responsible for providing the correct alignment, in the latter case an incorrectly aligned slice can be handled by taking a subslice like @Soveu suggested.

Soveu, 2021-03-19:
> Alternatively, a new type could be defined to represent a possibly uninitialized buffer with certain alignment requirements, e.g.:
> 
> ```rust
> #[repr(C, align(8))]
> pub struct MemoryMapBuffer([MaybeUninit<u8>]);
> ```

`align(8)` also requires that length is a multiple of 8, so really this is just `[u64]` (assuming u64 is aligned to 8)
[Otherwise it is UB according to Miri](https://play.rust-lang.org/?version=nightly&mode=release&edition=2018&gist=ee9fdb1da4e96e5ea11cf715e6dccbcc)

hanmertens, 2021-03-19:
Minor nitpick: it's not that the length has to be a multiple of 8, because changing `let newlen = newlen - (newlen % align);` to `let newlen = newlen - 8;` [stops Miri from complaining as well](https://play.rust-lang.org/?version=nightly&mode=release&edition=2018&gist=beb27dbcd41d5b83eb7b81eacdefedf6). I think the UB is caused by the `Vec` allocation not backing the buffer's length rounded up to a multiple of 8 (although I don't understand why that would be UB). In any case, just using a slice is probably a better solution then.

Soveu, 2021-03-19:
It is UB, because `size_of_val % align_of_val` must be 0
In this case, the wrapper struct has size in multiple of 8, even though the inner slice can have any size (so a multiple of 1)

phip1611, 2024-06-23:
> MaybeUninit<u8>

I think this doesn't  make sense. Every possible memory value is a valid u8. MaybeUninit is only needed for sophisticated types IMHO.

I'm closing this due to inactivity. Please re-open with more specific pointers and suggestions matching the current code-base if necessary.

ping @nicholasbishop

closed 2024-06-23 by phip1611

kotauskas, 2024-06-23:
> Every possible memory value is a valid `u8`.

@phip1611 Due to the value of an uninitialized byte being considered "floating" (unfrozen) by the compiler, [it is UB to read an uninitialized integer](<https://doc.rust-lang.org/std/mem/union.MaybeUninit.html#initialization-invariant>) despite every bit pattern being valid.

nicholasbishop, 2024-06-23:
I agree this is fine to close, I think we've resolved the places where UB due to unalignment was possible. (If more such places are found, we can create a fresh bug.)

> Every possible memory value is a valid u8. MaybeUninit is only needed for sophisticated types IMHO, such as MaybeUninit<MyCoolStruct>.

This isn't correct unfortunately due to the way optimizing compilers work. There are probably more places in uefi-rs where it would be good to use `MaybeUninit`, but again we can open fresh or PRs if we find such places.) https://www.ralfj.de/blog/2019/07/14/uninit.html is a good explanation of why it's not sufficient to just think about the physical hardware value.

phip1611, 2024-06-24:
Thanks for the blog post - very interesting!