rust-osdev/uefi-rs #155: The usage of the iterator returned from `exit_boot_services`

toku-sa-n, 2020-08-05:
Thanks to [this comment](https://github.com/rust-osdev/uefi-rs/pull/151#issuecomment-667876158), now I know that iterators can be converted into `Vec` using `.copied().collect()`. However, this can't be applied to the iterator returned from `exit_boot_services` as the function disables the global allocator.

So what's the intended usage of the iterator returned from `exit_boot_services`? I think without converting this iterator into like a slice, it will be difficult to treat this.

GabrielMajeri, 2020-08-05:
It disables the global allocator based on the _UEFI memory allocation_ functions, but at that point the _kernel's memory allocator_ is supposed to take over.

In fact, this is a cat and mouse problem: if you were allowed to `collect` the returned iterator, it would change the memory map. If it changes the memory map, you'd have to get and `collect` the memory map again, which would change it again and so on.

The best solution is to use the returned iterator to identify a large enough area of memory to initialize the kernel memory allocator and then use it to store the memory map (you don't have to store it the way UEFI returned it, you can convert it to your kernel's memory map format).

toku-sa-n, 2020-08-05:
Thanks for useful information.

>You can also check out the way Linux does it, which is still pretty convoluted: https://lwn.net/Articles/632528/

I'll check it. Thanks!

closed 2020-08-05 by toku-sa-n