rust-osdev/uefi-rs #172: why i get a '\r' not a '\n'>? when i use the text.input

zzhgithub, 2020-10-03:
why i get a '\r' not a '\n'>? when i use the text.input.
how can i get a '\n' when i `enter`?

thanks .
```
pub async fn test_input(st: SystemTable<Boot>) {
    loop {
        let events = &mut [st.stdin().wait_for_key_event()];
        &st.boot_services().wait_for_event(events).unwrap();
        match st.stdin().read_key().unwrap().unwrap() {
            Some(key) => {
                match key {
                    uefi::proto::console::text::Key::Printable(char16) => {
                        // 为什么这里的回车\r 不是 \n 呢？
// here !!!!
                        write!(st.stdout(), "{}", char16).unwrap();
                    }
                    uefi::proto::console::text::Key::Special(scanCode) => {
                        info!("{:?}", key)
                    }
                }
            }
            None => {}
        };
    }
}
```

GabrielMajeri, 2020-10-04:
Hello, I've tested and reproduced this locally, but I don't think there's something we can fix.

This is just the way UEFI implementation works. UEFI is heavily inspired by Windows, and an `enter` on Windows would mean `\r\n` (I'm guessing UEFI returns only the first `\r`).

zzhgithub, 2020-10-04:
Thank you for your response.
I understand. By the way, is there a good way to determine what the input value is? I find that what is returned is a structure called char16. which I can't easily access. Or can I just convert the data to u8 and let something like `pc-keyboard` cargo handle it? Do you have any good suggestions?

GabrielMajeri, 2020-10-05:
The `Char16` structure can be converted with `.into()` to a `u16` or a `char`, which you can then use normally.

nicholasbishop, 2021-12-04:
Closing as it sounds like the behavior is expected.

closed 2021-12-04 by nicholasbishop