rust-osdev/uefi-rs #174: SetVirtualAddressMap hangs

jess-sch, 2020-10-07:
I'm not quite sure why this is, but when I write the following code, it always hangs at the call to `set_virtual_address_map`. It does not panic (which I know because the panic handler is set to immediately reboot), it just hangs there forever doing nothing.

I've also tried this with `allocate_pages` (properly aligned with ptr::align_offset) instead of on the stack, but that doesn't help either.

```
#[no_mangle]
unsafe extern "efiapi" fn efi_main(image: Handle, st: SystemTable<Boot>) -> Status {
    {
        let mut map: MaybeUninit<[MemoryDescriptor; 4096]> = MaybeUninit::uninit();
        let (st, iter) = st
            .exit_boot_services(
                image,
                core::slice::from_raw_parts_mut(map.as_mut_ptr() as *mut u8, sizeof(&map)),
            )?
            .split()
            .1;
        // Store a reference to the runtime services so our panic handler can use it to reboot
        RUNTIME_SERVICES.store(
            st.runtime_services() as *const RuntimeServices as *mut RuntimeServices,
            AtomicOrdering::SeqCst,
        );
        let mut map =
            core::slice::from_raw_parts_mut(map.as_mut_ptr() as *mut MemoryDescriptor, iter.len());
        //for md in map.iter_mut() {
        //    md.virt_start = md.phys_start;
        //}
        let _ = st.runtime_services().set_virtual_address_map(map);
        st.runtime_services().reset(ResetType::Shutdown, Status::SUCCESS, None);
    };
    crate::main()
}

```

I'm executing the binary on Fedora by running it in a UEFI shell in QEMU:
> qemu-system-x86_64 -enable-kvm -bios /usr/share/edk2/ovmf/OVMF_CODE.fd -m 2G -drive file=fat:rw:target/x86_64/debug -net none

GabrielMajeri, 2020-10-08:
The only cause I could think of is that you're declaring a _contiguous_ array of `MemoryDescriptor`; there's no guarantee in the spec that the returned memory map is an array with no gaps, this is why you're given an iterator, and why the buffer should be bigger than required.

You could try collecting the given iterator into a newly allocated buffer, and use that for `set_virtual_address_map`.

jess-sch, 2020-10-08:
> The only cause I could think of is that you're declaring a _contiguous_ array of `MemoryDescriptor`; there's no guarantee in the spec that the returned memory map is an array with no gaps, this is [...]

Thanks for the suggestion; unfortunately collecting the iterator into a separate buffer does not improve the situation at all.

jess-sch, 2020-10-12:
Well, now I feel a little stupid. I've changed it up a little (not entirely sure what was wrong with my previous implementation), but now it works. Sorry for bothering you.

```
use core::sync::atomic::{AtomicPtr, AtomicUsize, Ordering as AtomicOrdering};
use uefi::{
    prelude::*,
    table::{
        boot::{AllocateType, MemoryDescriptor, MemoryType},
        cfg::ConfigTableEntry,
    },
};

static RUNTIME_SERVICES: AtomicPtr<RuntimeServices> = AtomicPtr::new(core::ptr::null_mut());
pub fn runtime_services() -> &'static RuntimeServices {
    unsafe { &*RUNTIME_SERVICES.load(AtomicOrdering::Relaxed) }
}

static CONFIG_TABLES: AtomicPtr<ConfigTableEntry> = AtomicPtr::new(core::ptr::null_mut());
static CONFIG_TABLES_LEN: AtomicUsize = AtomicUsize::new(0);
pub fn config_tables() -> &'static [ConfigTableEntry] {
    unsafe {
        let len = CONFIG_TABLES_LEN.load(AtomicOrdering::Relaxed);
        let data = &*CONFIG_TABLES.load(AtomicOrdering::Relaxed) as *const ConfigTableEntry;
        &*core::ptr::slice_from_raw_parts(data, len)
    }
}

#[no_mangle]
unsafe extern "efiapi" fn efi_main(image: Handle, st: SystemTable<Boot>) -> Status {
    {
        let count = (st.boot_services().memory_map_size() / 4096) + 1;
        let src = st
            .boot_services()
            .allocate_pages(AllocateType::AnyPages, MemoryType::LOADER_DATA, count)?
            .split()
            .1 as *mut u8;
        let tgt = st
            .boot_services()
            .allocate_pages(AllocateType::AnyPages, MemoryType::LOADER_DATA, count)?
            .split()
            .1 as *mut u8;
        let tgt = tgt.add(core::mem::size_of::<MemoryDescriptor>()) as *mut MemoryDescriptor;
        let (st, iter) = st
            .exit_boot_services(
                image,
                &mut *core::ptr::slice_from_raw_parts_mut(src, count * 4096),
            )?
            .split()
            .1;
        {
            let step = core::mem::size_of::<MemoryDescriptor>();
            let mut cur = tgt;
            for &md in iter {
                let mut md = md;
                cur.write(md);
                cur = cur.add(step);
            }
            let mut tgt =
                core::slice::from_raw_parts_mut(tgt, (cur as usize - tgt as usize) / step);
            let _ = st
                .runtime_services()
                .set_virtual_address_map(&mut tgt)
                .unwrap();
        }
        RUNTIME_SERVICES.store(
            st.runtime_services() as *const RuntimeServices as *mut RuntimeServices,
            AtomicOrdering::Relaxed,
        );
        CONFIG_TABLES.store(
            st.config_table() as *const _ as *mut _,
            AtomicOrdering::Relaxed,
        );
        CONFIG_TABLES_LEN.store(st.config_table().len(), AtomicOrdering::Relaxed);
    };
    crate::main()
}
```

closed 2020-10-12 by jess-sch