rust-osdev/uefi-rs #175: Have SimpleFileSystem's ``open_volume`` function  iterate and select the boot device

Swampman08, 2020-10-11:
Hi there, I have been using the ``uefi-rs`` library for some time now and it has worked perfectly, except for 1 issue:

On most devices, the SimpleFileSystem protocol's default volume is the boot volume, which is what you would want as any files your UEFI app may want to use should be on the boot volume, however, on some devices (particularly laptops) the default volume is not set to the boot volume meaning it is not possible to access any files on the boot volume through the  ``open_volume`` function.

Would it be possible to iterate through all the volumes on a device to find the boot volume instead of just choosing the SimpleFileSystem protocol's default volume?

Thank you for your time.

Swampman08, 2020-10-12:
As it turns out, this issue is something I can fix using the ``locate_handle`` function

closed 2020-10-12 by Swampman08

sreehax, 2021-02-28:
Just in case anyone else is stuck, I also spent a ton of time with trial and error trying to figure out how to open the boot volume in SimpleFileSystem, but I figured it out eventually. Here was my procedure:
1) use handle_protocol with the LoadedImage protocol on the image handle you get from the uefi entry point. Then you can call .device() to get a handle to the storage device the image was booted from
2) use handle_protocol with the DevicePath protocol on the storage device handler from step 1 to get the boot DevicePath
3) use locate_device_path with the SimpleFileSystem protocol and the DevicePath from step 2 to get a handle to the file system itself
4) finally use handle_protocol with the SimpleFileSystem protocol and the file system handle from step 3 to get a SimpleFileSystem instance to the boot volume!

GabrielMajeri, 2021-02-28:
@sreehax Thanks for describing the procedure! I've created PR #201, which adds a helper method providing this functionality.