rust-osdev/uefi-rs #176: Error: ‘if' is not allowed in a 'const fn' when compiling

dingiso, 2020-10-15:
I'm sure that it's my problem but,I don't know how to fix it. The problem is When compiling , it said :
```
error[E0658]: `if` is not allowed in a `const fn`
   --> /home/dingisoul/.cargo/registry/src/github.com-1ecc6299db9ec823/uefi-0.5.0/src/table/boot.rs:653:9
    |
653 |         assert!(value >= 0x80000000);
    |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |
    = note: see issue #49146 <https://github.com/rust-lang/rust/issues/49146> for more information
    = help: add `#![feature(const_if_match)]` to the crate attributes to enable
    = note: this error originates in a macro (in Nightly builds, run with -Z macro-backtrace for more info)

error: aborting due to previous error

For more information about this error, try `rustc --explain E0658`.
error: could not compile `uefi`.
```
But I don't know where to add the feature ,  or there is something wrong with my rustc version ? 
```
rustc 1.45.0-nightly (56daaf669 2020-06-03)

```

josephlr, 2020-10-15:
You need to update your compiler to the latest nightly

dingiso, 2020-10-15:
Thank you , I'll try it now, and see what happened

dingiso, 2020-10-16:
It's the wrong of Rustup version , you have to update the nightly version and change the Rust-tool_chain

closed 2020-10-16 by dingiso