rust-osdev/uefi-rs #178: Linking with rust-lld failed

ahmtcn123, 2020-10-22:
```
Caused by:
  process didn't exit successfully: `rustc --crate-name bos --edition=2018 src/main.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 -C metadata=43dafc72fc824056 -C extra-filename=-43dafc72fc824056 --out-dir /mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps --target x86_64-unknown-uefi -C incremental=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/incremental -L dependency=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps -L dependency=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/debug/deps --extern 'noprelude:alloc=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/liballoc-6564cef7201a043b.rlib' --extern 'noprelude:compiler_builtins=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/libcompiler_builtins-a908511f89d37fab.rlib' --extern 'noprelude:core=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/libcore-a5ca24864acc80ba.rlib' --extern log=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/liblog-a23a0c62874bf5ea.rlib --extern rlibc=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/librlibc-cea020b5530bff7d.rlib --extern uefi=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/libuefi-dad491a0a9497efa.rlib --extern uefi_services=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/libuefi_services-4c4948bb9b06fb95.rlib --extern x86_64=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/libx86_64-1c1095e404e57e60.rlib -Z unstable-options` (exit code: 1)
```

Also in trying in wsl ubuntu 20.04

phil-opp, 2020-10-22:
Could you post the full error output? I think you might have missed to include the actual error message.

ahmtcn123, 2020-10-22:
That output was in wsl. I captured windows output for now

```
warning: unused manifest key: package.rlibc
warning: unused manifest key: unstable     
   Compiling rustc-std-workspace-core v1.99.0 (C:\Users\ahmet\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\src\tools\rustc-std-workspace-core)
   Compiling backtrace-sys v0.1.37
   Compiling compiler_builtins v0.1.32
   Compiling libc v0.2.71
   Compiling alloc v0.0.0 (C:\Users\ahmet\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\src\liballoc)
   Compiling cfg-if v0.1.10
   Compiling rustc-demangle v0.1.16
   Compiling panic_abort v0.0.0 (C:\Users\ahmet\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\src\libpanic_abort)
The following warnings were emitted during compilation:

warning: src/libbacktrace/alloc.c:1:0: warning: -fPIC ignored for target (all code is position independent) [enabled by default]
warning:  /* alloc.c -- Memory allocation without mmap.
warning:  ^
warning: src/libbacktrace/dwarf.c:1:0: warning: -fPIC ignored for target (all code is position independent) [enabled by default]
warning:  /* dwarf.c -- Get file/line information from DWARF for backtraces.
warning:  ^
warning: src/libbacktrace/fileline.c:1:0: warning: -fPIC ignored for target (all code is position independent) [enabled by default]
warning:  /* fileline.c -- Get file and line number information in a backtrace.
warning:  ^
warning: src/libbacktrace/posix.c:1:0: warning: -fPIC ignored for target (all code is position independent) [enabled by default]
warning:  /* posix.c -- POSIX file I/O routines for the backtrace library.
warning:  ^
warning: src/libbacktrace/sort.c:1:0: warning: -fPIC ignored for target (all code is position independent) [enabled by default]
warning:  /* sort.c -- Sort without allocating memory
warning:  ^
warning: src/libbacktrace/state.c:1:0: warning: -fPIC ignored for target (all code is position independent) [enabled by default]
warning:  /* state.c -- Create the backtrace state.
warning:  ^
warning: src/libbacktrace/mmapio.c:1:0: warning: -fPIC ignored for target (all code is position independent) [enabled by default]
warning:  /* mmapio.c -- File views using mmap.
warning:  ^
warning: src/libbacktrace/mmapio.c:37:22: fatal error: sys/mman.h: No such file or directory
warning:  #include <sys/mman.h>
warning:                       ^
warning: compilation terminated.

error: failed to run custom build command for `backtrace-sys v0.1.37`

Caused by:
  process didn't exit successfully: `C:\Users\ahmet\Desktop\Projects\InBuild\bos\target\debug\build\backtrace-sys-d806ac0cdbe09e46\build-script-build` (exit code: 1)
  --- stdout
  cargo:rustc-cfg=rdos
  TARGET = Some("x86_64-unknown-uefi")
  OPT_LEVEL = Some("0")
  HOST = Some("x86_64-pc-windows-msvc")
  CC_x86_64-unknown-uefi = None
  CC_x86_64_unknown_uefi = None
  TARGET_CC = None
  CC = None
  CROSS_COMPILE = None
  CFLAGS_x86_64-unknown-uefi = None
  CFLAGS_x86_64_unknown_uefi = None
  TARGET_CFLAGS = None
  CFLAGS = None
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some("true")
  CARGO_CFG_TARGET_FEATURE = Some("fxsr")
  running: "cc" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-g" "-fno-omit-frame-pointer" "-m64" "-I" "src/libbacktrace" "-I" "C:\\Users\\ahmet\\Desktop\\Projects\\InBuild\\bos\\target\\x86_64-unknown-uefi\\debug\\build\\backtrace-sys-1e875274b97185e8\\out" "-fvisibility=hidden" "-DBACKTRACE_ELF_SIZE=64" "-DBACKTRACE_SUPPORTED=1" "-DBACKTRACE_USES_MALLOC=1" "-DBACKTRACE_SUPPORTS_THREADS=0" "-DBACKTRACE_SUPPORTS_DATA=0" "-DHAVE_DL_ITERATE_PHDR=1" "-D_GNU_SOURCE=1" "-D_LARGE_FILES=1" "-Dbacktrace_full=__rdos_backtrace_full" "-Dbacktrace_dwarf_add=__rdos_backtrace_dwarf_add" "-Dbacktrace_initialize=__rdos_backtrace_initialize" "-Dbacktrace_pcinfo=__rdos_backtrace_pcinfo" "-Dbacktrace_syminfo=__rdos_backtrace_syminfo" "-Dbacktrace_get_view=__rdos_backtrace_get_view" "-Dbacktrace_release_view=__rdos_backtrace_release_view" "-Dbacktrace_alloc=__rdos_backtrace_alloc" "-Dbacktrace_free=__rdos_backtrace_free" "-Dbacktrace_vector_finish=__rdos_backtrace_vector_finish" "-Dbacktrace_vector_grow=__rdos_backtrace_vector_grow" "-Dbacktrace_vector_release=__rdos_backtrace_vector_release" "-Dbacktrace_close=__rdos_backtrace_close" "-Dbacktrace_open=__rdos_backtrace_open" "-Dbacktrace_print=__rdos_backtrace_print" "-Dbacktrace_simple=__rdos_backtrace_simple" "-Dbacktrace_qsort=__rdos_backtrace_qsort" "-Dbacktrace_create_state=__rdos_backtrace_create_state" "-Dbacktrace_uncompress_zdebug=__rdos_backtrace_uncompress_zdebug" "-Dmacho_get_view=__rdos_macho_get_view" "-Dmacho_symbol_type_relevant=__rdos_macho_symbol_type_relevant" "-Dmacho_get_commands=__rdos_macho_get_commands" "-Dmacho_try_dsym=__rdos_macho_try_dsym" "-Dmacho_try_dwarf=__rdos_macho_try_dwarf" "-Dmacho_get_addr_range=__rdos_macho_get_addr_range" "-Dmacho_get_uuid=__rdos_macho_get_uuid" "-Dmacho_add=__rdos_macho_add" "-Dmacho_add_symtab=__rdos_macho_add_symtab" "-Dmacho_file_to_host_u64=__rdos_macho_file_to_host_u64" "-Dmacho_file_to_host_u32=__rdos_macho_file_to_host_u32" "-Dmacho_file_to_host_u16=__rdos_macho_file_to_host_u16" "-o" "C:\\Users\\ahmet\\Desktop\\Projects\\InBuild\\bos\\target\\x86_64-unknown-uefi\\debug\\build\\backtrace-sys-1e875274b97185e8\\out\\src/libbacktrace/alloc.o" "-c" "src/libbacktrace/alloc.c"
  cargo:warning=src/libbacktrace/alloc.c:1:0: warning: -fPIC ignored for target (all code is position independent) [enabled 
by default]
[... cut ...]
40 | pub unsafe extern "C" fn __rust_start_panic(_payload: usize) -> u32 {
   |                          ------------------                     ^^^ expected `u32`, found `()`
   |                          |
   |                          implicitly returns `()` as its body has no tail or `return` expression

error: aborting due to 2 previous errors

Some errors have detailed explanations: E0308, E0425.
For more information about an error, try `rustc --explain E0308`.
error: build failed
```

command:
```
cargo build -Z build-std --target x86_64-unknown-uefi
```

ahmtcn123, 2020-10-22:
Same command in wsl ubuntu
```
warning: unused manifest key: package.rlibc
warning: unused manifest key: unstable
       Fresh core v0.0.0 (/home/ahmtcn123/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
       Fresh rustc-std-workspace-core v1.99.0 (/home/ahmtcn123/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
       Fresh compiler_builtins v0.1.35
       Fresh alloc v0.0.0 (/home/ahmtcn123/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/alloc)
       Fresh ucs2 v0.1.0
       Fresh cfg-if v0.1.10
       Fresh bit_field v0.9.0
       Fresh os_bootinfo v0.2.1
       Fresh usize_conversions v0.2.0
       Fresh ux v0.1.3
       Fresh rlibc v1.0.0
       Fresh log v0.4.11
       Fresh bitflags v1.2.1
   Compiling uefi v0.1.0 (/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/uefi-rs)
   Compiling x86_64 v0.2.14
       Fresh x86_64 v0.12.2
     Running `rustc --crate-name uefi --edition=2018 uefi-rs/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 -C metadata=86b882a427391f40 -C extra-filename=-86b882a427391f40 --out-dir /mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps --target x86_64-unknown-uefi -C incremental=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/incremental -L dependency=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps -L dependency=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/debug/deps --extern 'noprelude:alloc=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/liballoc-6564cef7201a043b.rmeta' --extern bitflags=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/libbitflags-955383eba894e97a.rmeta --extern 'noprelude:compiler_builtins=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/libcompiler_builtins-a908511f89d37fab.rmeta' --extern 'noprelude:core=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/libcore-a5ca24864acc80ba.rmeta' --extern ucs2=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/libucs2-602da96ae24cb899.rmeta -Z unstable-options`
     Running `rustc --crate-name x86_64 /home/ahmtcn123/.cargo/registry/src/github.com-1ecc6299db9ec823/x86_64-0.2.14/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 -C metadata=8a1b2f6d9f3609cb -C extra-filename=-8a1b2f6d9f3609cb --out-dir /mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps --target x86_64-unknown-uefi -L dependency=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps -L dependency=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/debug/deps --extern 'noprelude:alloc=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/liballoc-6564cef7201a043b.rmeta' --extern bit_field=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/libbit_field-db2d1c58ee7ae8d3.rmeta --extern bitflags=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/libbitflags-955383eba894e97a.rmeta --extern 'noprelude:compiler_builtins=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/libcompiler_builtins-a908511f89d37fab.rmeta' --extern 'noprelude:core=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/libcore-a5ca24864acc80ba.rmeta' --extern os_bootinfo=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/libos_bootinfo-4ef733b72b2e0deb.rmeta --extern usize_conversions=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/libusize_conversions-c7538daaf466208b.rmeta --extern ux=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/libux-175f80d262671f69.rmeta -Z unstable-options --cap-lints allow`
error[E0658]: negative trait bounds are not yet fully implemented; use marker types for now
   --> uefi-rs/src/proto/macros.rs:32:14
    |
32  |           impl !Send for $p {}
    |                ^^^^^
    | 
   ::: uefi-rs/src/proto/console/gop.rs:271:1
    |
271 | / impl_proto! {
272 | |     protocol GraphicsOutput {
273 | |         GUID = 0x9042a9de, 0x23dc, 0x4a38, [0x96, 0xfb, 0x7a, 0xde, 0xd0, 0x80, 0x51, 0x6a];
274 | |     }
275 | | }
    | |_- in this macro invocation
    |
    = note: see issue #68318 <https://github.com/rust-lang/rust/issues/68318> for more information
    = help: add `#![feature(negative_impls)]` to the crate attributes to enable
    = note: this error originates in a macro (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0658]: negative trait bounds are not yet fully implemented; use marker types for now
   --> uefi-rs/src/proto/macros.rs:34:14
    |
34  |           impl !Sync for $p {}
    |                ^^^^^
    | 
   ::: uefi-rs/src/proto/console/gop.rs:271:1
    |
271 | / impl_proto! {
272 | |     protocol GraphicsOutput {
273 | |         GUID = 0x9042a9de, 0x23dc, 0x4a38, [0x96, 0xfb, 0x7a, 0xde, 0xd0, 0x80, 0x51, 0x6a];
274 | |     }
275 | | }
[... cut ...] op/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/libusize_conversions-c7538daaf466208b.rmeta --extern ux=/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/target/x86_64-unknown-uefi/debug/deps/libux-175f80d262671f69.rmeta -Z unstable-options --cap-lints allow` (exit code: 1)
warning: build failed, waiting for other jobs to finish...
error: aborting due to 14 previous errors; 3 warnings emitted

For more information about this error, try `rustc --explain E0658`.
error: build failed
```

phil-opp, 2020-10-22:
You seem to be using some very old crate versions. Also, are you sure that you are really using this `uefi` crate? It appears to be in a local folder (`/mnt/c/Users/ahmet/Desktop/Projects/InBuild/bos/uefi-rs`) and its version seems to be v0.1.0.

This crate is currently at version 0.6.0. Version 0.1.0 was originally released in 2016 and has been yanked since then.

ahmtcn123, 2020-10-25:
Changing version worked thank you for help!

closed 2020-10-25 by ahmtcn123