rust-osdev/uefi-rs #181: Use an automated parser to generate EFI service VMTs

stevefan1999-personal, 2020-10-29:
We could always parse the tables from EDK2, i.e. here:
https://github.com/tianocore/edk2/blob/f1567720b13a578ffa54716119f826df622babcd/MdePkg/Include/Uefi/UefiSpec.h

We are certainly not going to [immunant/c2rust](https://github.com/immunant/c2rust) cause it does not have enough information without pulling the entire EDK2 I think, so I say we could use a C parser in Rust. I'm investigating [vickenty/lang-c](https://github.com/vickenty/lang-c) for this.

The generate will generate a 1-1 correspondence C-to-Rust unsafe method (because having unsafe function pointer at struct level isn't enough, there's no UFCS for example IIRC). 

Once the generator is done, the current handwritten helper impls will be moved to a wrapped extension trait that you could import.

Rust compiler should be able to smartly deflate the VMT and directly transform the supposed virtual trait call into the equivalent raw EFI function pointer call as you usually do.

GabrielMajeri, 2020-10-30:
This is quite an undertaking, but would definitely boost our progress of porting the whole UEFI API significantly!

hussein-aitlahcen, 2021-01-26:
@stevefan1999-personal I would like to know if this if you have been able to investigate on this? I raised the same kind of issue with #195 and would like to give a shot to bindgen on EDK2 unless you are already working on something similar :).

phip1611, 2024-06-23:
I'm closing this due to inactivity. Please re-open with more specific pointers and suggestions matching the current code-base if necessary.

ping @nicholasbishop

closed 2024-06-23 by phip1611