rust-osdev/uefi-rs #184: Creating your own protocols

notzortex, 2020-11-29:
I've looked through previous issues and saw this has been mentioned before, but even with that pull request i struggle with creating my own protocols.
How am i supposed to do this? I see there's a guide under the contributing file, but doesn't seem to have any useful information outside of what i know already. Has there been any more work towards exposing the correct stuff?

GabrielMajeri, 2020-11-29:
Hello. Are you referring to creating your own novel protocol, or providing a Rust wrapper for an existing UEFI protocol? If so, which protocol are you interested in implementing?

notzortex, 2020-11-29:
I was referring to providing a rust wrapper for an existing uefi protocol.
My goal is to implement pxe protocols from the efi crate (to get networking) which from what i'm seeing is abandoned which is why i chose to use this crate

GabrielMajeri, 2020-11-29:
I don't know if there's a very well documented process, since this is mostly reverse engineering [the UEFI spec](https://www.uefi.org/sites/default/files/resources/UEFI%20Spec%202_6.pdf) :sweat_smile:

The steps I'd follow are:
- Pick a protocol
[image]
- Define it
```rust
#[repr(C)]
#[unsafe_guid("A19832B9-...")]
#[derive(Protocol)]
pub struct SimpleNetworkProtocol {
    revision: u64,
    // ...
}
```
- Define the function prototypes and wrap them. You can find some example functions in the [`SimpleTextOutputProtocol`](https://github.com/rust-osdev/uefi-rs/blob/master/src/proto/console/text/output.rs)

You can define as many functions/types as required, based on what you're trying to do in your code.

notzortex, 2020-11-29:
This much i understood (and had already done) but i'm getting issues with defining the actual struct.
```
error[E0433]: failed to resolve: unresolved import
  --> src\main.rs:57:10
   |
57 | #[derive(Protocol)]
   |          ^^^^^^^^
   |          |
   |          unresolved import
   |          help: a similar path exists: `uefi::proto`
```
I think i'm including the right things, i've been messing around with it for a while now without any results.
This is probably something im doing wrong, so any help is appreciated.

GabrielMajeri, 2020-11-29:
Are you trying to define the `Protocol` in your binary? You might have to add the `uefi-macros` crate as a dependency to get the required derive.

The easier/recommended alternative is to fork the crate, add the protocols directly to `uefi-rs`, and then submit a PR with the changes :)

notzortex, 2020-11-29:
I tried adding uefi-macros without much success (same errors)

i was doing it like this for testing anyways, forking and making a PR should be easier.

closed 2020-11-30 by notzortex