rust-osdev/uefi-rs #195: Boot services implementation

hussein-aitlahcen, 2021-01-26:
Hi folks,

First I would like yo thank you for this amazing lib.
I tried to use the library and figured out that for instance, the `load_image` and `unload_image` services are not implemented?
Do we have a rationale behind all the `field: usize` in the BootServices table or are they just default until someone implement them? Can't we provide the raw and unsafe functions to everyone, allowing people to choose between directly calling them or using the type safe api?

Thanks

GabrielMajeri, 2021-01-26:
Hello! As you've noted, those functions aren't implemented _yet_. The UEFI spec is really big and this project has grown on an as-needed basis; when people wanted to use a certain function, they made a PR to add it to `uefi-rs` :)

The `function: usize` fields in the tables are there to provide the correct padding and alignment, at least until all the functions will be supported.

Providing raw and unsafe functions directly: we could make them public and hide them from the docs, although I don't know how beneficial such a change would be. The whole point of this crate is to provide easy and safe access to all of UEFI's functionality (although it's still a work-in-progress).

If you have the time, it should be fairly easy to fork this crate, and make a pull request with the `load/unload_image` functions. Let me know if you need any help with the code :D

hussein-aitlahcen, 2021-01-26:
> Hello! As you've noted, those functions aren't implemented _yet_. The UEFI spec is really big and this project has grown on an as-needed basis; when people wanted to use a certain function, they [...]

Thanks for the quick answer. Regarding making the fields public, I meant that perhaps it's a good thing until every service has a more abstract and type safe layer :). Anyway, I will definitely fork and will try to implement them!

hussein-aitlahcen, 2021-01-26:
@GabrielMajeri any thought on trying to use the bindgen package to generate the UEFI spec from the EDK2 repo and the wrap them? I have successfully generated the spec, available [here](https://gist.github.com/hussein-aitlahcen/f977de62be239ad5cb57f1fff496b183). This generation can be platform-specific and done at compile time.

GabrielMajeri, 2021-01-26:
@hussein-aitlahcen Somebody started a discussion with a similar suggestion in issue #181. I'm not against such a change, it simply requires quite a bit of effort to refactor the existing code. I don't actively develop the crate anymore, I just maintain it (by reviewing and merging PRs and making sure it compiles :D). Contributions towards such a refactoring are welcome (as long as we don't break the current code).

We could have a separate `uefi-sys` crate, which exposes the raw, unsafe UEFI API (a [pattern](https://kornel.ski/rust-sys-crate) used in many other crates providing bindings to C APIs).

GabrielMajeri, 2022-02-05:
Closing this issue since it has been superseded by #181 and had some work done towards it in #196.

closed 2022-02-05 by GabrielMajeri