rust-osdev/uefi-rs #197: New version of the `uefi-services` crate?

toku-sa-n, 2021-02-15:
v0.4.0 of the `uefi-services` crate can't be compiled because of the error of v0.12.4 of the `x86_64` crate.
```
error[E0557]: feature has been removed
 --> /home/runner/.cargo/registry/src/github.com-1ecc6299db9ec823/x86_64-0.12.4/src/lib.rs:9:43
  |
9 | #![cfg_attr(feature = "const_fn", feature(const_in_array_repeat_expressions))]
  |                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ feature has been removed
  |
  = note: removed due to causing promotable bugs
```
The latest version of the `x86_64` crate is 0.13.2 which solves the problem.
Is there a plan to publish a new version of the crate with the updated version of the `x86_64` crate?

GabrielMajeri, 2021-02-15:
Will push a new version as soon as #198 gets merged.

toku-sa-n, 2021-02-17:
Thank you for publishing the new version.

closed 2021-02-17 by toku-sa-n