rust-osdev/uefi-rs #206: RSDP + devices

genderev, 2021-04-27:
1. Why and when is RSDP not reported? If parsing ACPI is important to your crate, would exporting (in the BootInfo struct) specific entries from ACPI be more reliable than exporting RSDP as a whole?

2. What is the purpose of the line `let gop = unsafe { &mut *gop.get() };` ?

phil-opp, 2021-04-27:
To get the RSDP, you can use:

```rust
use uefi::table::cfg;

let mut config_entries = system_table.config_table().iter();
let rsdp_addr = config_entries
    .find(|entry| matches!(entry.guid, cfg::ACPI_GUID | cfg::ACPI2_GUID))
    .map(|entry| entry.address);
```

genderev, 2021-04-27:
Hi Phil, thank you for replying.

I should have asked this question in the bootloader repo.

closed 2021-04-27 by genderev

phil-opp, 2021-04-27:
No worries!

(Cross linking: the bootloader issue is at https://github.com/rust-osdev/bootloader/issues/152)