rust-osdev/uefi-rs #208: Can't build `uefi-services v0.5.0` with the latest Nightly

toku-sa-n, 2021-05-07:
It depends on `x86_64 v0.13.6`, which can't compile with the latest Nightly because of missing the `const_fn_trait_bound` feature gate. Could you publish the new version of the crate?

phil-opp, 2021-05-07:
Perhaps I'm missing something, but this crate doesn't seem to depend on the `x86_64` crate..

toku-sa-n, 2021-05-07:
> Perhaps I'm missing something, but this crate doesn't seem to depend on the `x86_64` crate..

Currently, yes, the crate doesn't depend on it. However, [`v0.5.0` actually depends on](https://crates.io/crates/uefi-services/0.5.0/dependencies).

phil-opp, 2021-05-07:
Ah, makes sense, thanks for clarifying!

GabrielMajeri, 2021-05-08:
I've published version `0.5.1` of the `uefi-services` crate, should fix the issue.