rust-osdev/uefi-rs #215: UEFI-rs alloc feature ?

JakubGawron23, 2021-05-12:
I was reviewing the 3rd edition of @phill-opp's blog_os and wanted to build his uefi application but stopped at ```uefi = { version = "0.8.0", features = ["alloc"] }``` because **cargo** can't find that function. And is it available in **uefi-rs** or not ?
```#![feature(alloc_error_handler)]

// link the alloc crate
extern crate alloc;

use alloc::vec::Vec;

#[entry]
fn efi_main(
    image: uefi::Handle,
    system_table: uefi::table::SystemTable<uefi::table::Boot>,
) -> uefi::Status {
    // ... (as before)

    // initialize the allocator
    unsafe {
        uefi::alloc::init(system_table.boot_services());
    }

    // we can now use the allocator
    let mut v = Vec::new();
    v.push(1);
    v.push(2);
    writeln!(stdout, "v = {:?}", v).unwrap();

    loop {}
}

/// This function is called when an allocation fails,
/// typically because the system is out of memory.
#[alloc_error_handler]
fn alloc_error(_layout: Layout) -> ! {
    panic!("out of memory")
}```

toku-sa-n, 2021-05-12:
Use `uefi-services` crate:
```toml
uefi = { version = "0.8.1", features = ["logger"] }
uefi-services = "0.5.1"
```

And initialize with:
```rust
fn init_uefi_utils(system_table: &SystemTable<Boot>) {
    uefi_services::init(system_table).expect_success("Failed to initialize_uefi_utilities");
}
```

toku-sa-n, 2021-05-12:
Ah, I found that `uefi` crate itself provides `alloc` feature, sorry.

JakubGawron23, 2021-05-12:
It still doesn't work and gives the error ```can't find crate for `alloc` ```

toku-sa-n, 2021-05-12:
Do you have these options in `.cargo/config.toml`?
```toml
[unstable]
build-std = ["core", "compiler_builtins", "alloc"]
build-std-features = ["compiler-builtins-mem"]
```
Or
```
cargo build -Z build-std=core,compiler_builtins,alloc -Z build-std-features=compiler-builtins-mem
```

JakubGawron23, 2021-05-12:
Well no, I forgot to add ```alloc```. Sorry for my stupidity

closed 2021-05-12 by JakubGawron23