rust-osdev/uefi-rs #218: How to load the kernel from a UEFI application

JakubGawron23, 2021-05-13:
I read the 3rd edition post about the UEFI app created by @phil-opp and would like to load a kernel build from the UEFI app.
I know that I need to parse the ELF file with **xmas-elf** and then use the assembler to call the main kernel function. Despite my attempts I still did not get this. And I wanted to ask if you know any example UEFI application which would load kernel as expected and it would work. Look forward to your reply 😊

GabrielMajeri, 2021-05-13:
As far as I can tell, [this repo](https://github.com/redox-os/bootloader-efi) is the relevant bootloader code from the Redox OS project. They don't use `uefi-rs`, instead they have their own UEFI wrapper, but the flow is similar.

nicholasbishop, 2022-01-22:
If you are looking to run the Linux kernel specifically, here's an example: https://chromium.googlesource.com/crdyboot/+/ff7f67b3b13824ba2badeb2be43467d726620429/crdyboot/src/linux.rs#119

The basic idea is to build Linux with the EFI stub, which makes the kernel look like a simple PE/COFF executable. Then you can parse the PE header to get the entry point function and call it. (The same basic idea should work with other UEFI-compatible PE executable, but you might need to add support for relocations.)

Hopefully this example and/or the redox example answer your question, feel free to reopen this issue if not.

closed 2022-01-22 by nicholasbishop