rust-osdev/uefi-rs #219: Proper keyboard interface?

netthier, 2021-05-13:
Hey, I'm not sure if this is possible in UEFI, but a proper keyboard interface would be really useful.
What I mean by that is the ability to check if a certain key is pressed at any time, instead of the queue-based approach currently implemented with the Input protocol.
Currently, when looping over read_key(), the method will return None most of the time.
It would be more useful if I could get a slice of all keys currently pressed.

GabrielMajeri, 2021-05-13:
Hello. Unfortunately, this is not the way UEFI (and by extension, this crate) is meant to be used. It's designed to be a low-level environment which provides you just enough functions to be able to boot an OS kernel and load the required drivers. Therefore, if you need more advanced keyboard input, you'll have to write a driver for the specific device you're targeting.

For example, if the keyboard is an USB device, you're expected to write an USB keyboard driver and directly read its state (using polling at a high frequency, since [that's how USB devices work](https://stackoverflow.com/a/4813541/5723188)).

josephlr, 2021-05-14:
Right now [`Input` is a direct wrapper](https://github.com/rust-osdev/uefi-rs/blob/68d21a660e2d005159fd8fdabfdf5c26d764e305/src/proto/console/text/input.rs#L5-L13) around [`SIMPLE_TEXT_INPUT_PROTOCOL`](https://github.com/tianocore/edk2/blob/master/MdePkg/Include/Protocol/SimpleTextIn.h), so (as @GabrielMajeri said) it inherits the same queue-based approach from that protocol. 

However, UEFI also has [`EFI_SIMPLE_TEXT_INPUT_EX_PROTOCOL`](https://github.com/tianocore/edk2/blob/master/MdePkg/Include/Protocol/SimpleTextInEx.h) which has a better `read_key()` function that provides additional [`KeyShiftState` and `KeyToggleState`](https://github.com/tianocore/edk2/blob/master/MdePkg/Include/Protocol/SimpleTextInEx.h#L64-L79) along with the optional [`Key`](https://docs.rs/uefi/0.9.0/uefi/proto/console/text/enum.Key.html). We could expose this protocol as well, which might help.

> Currently, when looping over read_key(), the method will return None most of the time.

This doesn't exactly solve your problem, but you could use the [`wait_for_key_event()`](https://docs.rs/uefi/0.9.0/uefi/proto/console/text/struct.Input.html#method.wait_for_key_event) method to avoid any need for looping.

netthier, 2021-05-14:
Thank you very much for both of your answers. The EX_PROTOCOL definitely looks promising.

GabrielMajeri, 2022-02-05:
Closing this issue since the original question has been reasonably answered

closed 2022-02-05 by GabrielMajeri