rust-osdev/uefi-rs #232: Rename or change the function signature of `memset` in boot services table

iankronquist, 2021-06-03:
In boot.rs we see there is an unsafe function named memset. This function has a signature different than memset(2), but has a name different than that in the edk2 sources. This is confusing and will lead to bugs in a very unsafe function when people mix up the order of the value and the size.

We should either name the function memset and give it the signature `memset(&self, buffer: *mut u8, value: u8, size: usize) `, or name the function set_mem and give it the signature from EFI_SET_MEM.
```
  pub unsafe fn memset(&self, buffer: *mut u8, size: usize, value: u8) {        (self.set_mem)(buffer, size, value);    }```

https://github.com/rust-osdev/uefi-rs/blob/948464c908aa4332e5ab224cc96880a512833bc9/src/table/boot.rs#L529

iankronquist, 2021-06-03:
I'm happy to make the change. We can actually do both, have a function named memset with the C stdlib signature and have one named set_mem with the uefi signature. This is a breaking change and should come with a point release imo.

GabrielMajeri, 2021-06-04:
@iankronquist Your help is definitely appreciated. We can make a new point release afterwards.

GabrielMajeri, 2021-06-07:
Fixed by #235

closed 2021-06-07 by GabrielMajeri