rust-osdev/uefi-rs #252: The  BlockIOMedia.block_size() return zero

chhdao, 2021-07-06:
I get some wrong data when I read disk block.
It output that the block size is 0 and  panic when reading (because of MEDIA_CHANGED). Test in qemu and physical computer.
How can I use BlockIO right?

Thanks.

```rust
/// Read the Disk
use crate::IMAGE_SYSTEM_TABLE;
use alloc::boxed::Box;
use alloc::vec;
use core::convert::TryInto;
use core::{u128, u32, u64};
use uefi::proto::media::block::BlockIO;

/// Get BlockIO Handle
unsafe fn get_disk() -> &'static mut uefi::proto::media::block::BlockIO {
    // get boot_services
    let boot_services = IMAGE_SYSTEM_TABLE.as_ref().unwrap().boot_services();

    // locate BlockIO protocol
    let block = match boot_services.locate_protocol::<BlockIO>() {
        Ok(some) => some.unwrap().get(),

        Err(_err) => panic!("Load BlockIO failed down!"),
    };

    crate::tool::print_fmt(format_args!("Get BlockIO done.\n"));

    block.as_mut().unwrap()
}

///  Get the GPT Header
unsafe fn get_gpt_header(disk: &mut BlockIO) -> Box<GptHeaderInfo> {
    let media = disk.media();

    // Prepare the block size
    let block_size = media.block_size();

   // =========
   // Will output 0
   // ==========
    crate::tool::print_fmt(format_args!("Block size:{0}\n", block_size));

    // prepare the buffer
    let mut block_buffer = vec![0u8; block_size.try_into().unwrap()].into_boxed_slice();
    
    // ============
    // Read the disk block
    // Will panic because of MEDIA_CHANGED
    // ============
    disk.read_blocks(media.media_id(), 1, &mut block_buffer)
        .unwrap()
        .unwrap();

    // panic before here
   unreachable!("panic before here");
}

/// this function will be called
pub fn get_partition(){
    unsafe {
        // Get disk handle
        let disk: &mut BlockIO = get_disk();

        // Read header
        let header_info = get_gpt_header(disk);
       
        // won't output this
        crate::tool::print_fmt(format_args!("Get GPT header done!\n"));
    }
}
```

nicholasbishop, 2021-07-06:
This sounds like the issue that was fixed by https://github.com/rust-osdev/uefi-rs/pull/234. That PR is merged but I think not yet in a released version of uefi-rs. You can try building against the master branch and see if that works for you.

chhdao, 2021-07-07:
It works on the master. Thanks.

closed 2021-07-07 by chhdao