rust-osdev/uefi-rs #259: Failed to run tests for aarch64

Andy-Python-Programmer, 2021-07-20:
Failed to run tests for aarch64 with: 

Command: `./build.py --ci --headless run --target aarch64`
Error: `Subprocess qemu-system-aarch64 exited with error code -11`

It does look like all of the tests have passed, its just the error code which is wierd.

## Edits

I added the `--verbose` flag to the `build.py` script to get a better overview of the build.

## Rust Version

```
rustup 1.24.3 (ce5817a94 2021-05-31)
info: This is the version for the rustup toolchain manager, not the rustc compiler.
info: The currently active `rustc` version is `rustc 1.55.0-nightly (014026d1a 2021-07-19)
````

## Output

[output.txt](https://github.com/rust-osdev/uefi-rs/files/6846558/stdout.txt) (output too big maximum for issue comment is 65536 characters)

GabrielMajeri, 2021-07-22:
I've tried running the command above on the latest `master`, with `rustc 1.55.0-nightly (32c9b7b09 2021-07-21)`, and it completed without errors.

Which version/build of OVMF are you using as EFI firmware for AArch64?

Andy-Python-Programmer, 2021-07-23:
I got it from: https://www.kraxel.org/repos/jenkins/edk2/edk2.git-aarch64-0-20210421.18.g15ee7b7689.noarch.rpm as https://github.com/rust-osdev/ovmf-prebuilt is built for x86_64

nicholasbishop, 2021-12-04:
-11 is a segfault, so I think most likely you are hitting a Qemu bug. I wasn't able to reproduce the crash with Qemu 6.1.0. If you are still hitting this segfault, please try a newer version of Qemu. If that still doesn't work, you might need to file a bug in the [Qemu bug tracker](https://gitlab.com/qemu-project/qemu/-/issues).

GabrielMajeri, 2022-02-05:
I've successfully ran `cargo xtask run --target aarch64 --headless --ci`, with the latest OVMF from Gerd Hoffman's RPM builds and `qemu-system-aarch64` version `6.0.0 (Debian 1:6.0+dfsg-2expubuntu1.1)`.

I'm going to close this issue for now since I've been unable to reproduce the problem.

However, I've noticed that the OVMF firmware prints quite a lot of debugging messages to the console; @Andy-Python-Programmer if this is still a problem for you, could you try also attaching that output as a text file? Maybe we'll be able to figure out what's going on from the logs.

closed 2022-02-05 by GabrielMajeri