rust-osdev/uefi-rs #262: Failed to build test-runner isolated.

CraftyDH, 2021-07-28:
## Failed to build test-runner isolated from the main repo.
I modified cargo.toml to use the git versions of uefi and uefi-services instead of a local path.

ie:
```toml
[dependencies]
uefi = { path = "..", features = ['exts'] }
uefi-services = { path = "../uefi-services" }
```
was changed to
```toml
[dependencies]
uefi = { git = "https://github.com/rust-osdev/uefi-rs", features = ['exts'] }
uefi-services = { git = "https://github.com/rust-osdev/uefi-rs" }
```

However the following errors were generated.
Is their any reason for why this change should not work?

### Build log
```shell
C:/uefi-rs/uefi-test-runner> cargo +nightly build --target x86_64-unknown-uefi
   Compiling uefi-test-runner v0.2.0 (C:\Users\Daniel\Documents\GitHub\uefi-rs\uefi-test-runner)
error[E0308]: mismatched types
  --> uefi-test-runner\src\main.rs:26:25
   |
26 |     uefi_services::init(&mut st).expect_success("Failed to initialize utilities");
   |                         ^^^^^^^ expected struct `uefi::table::system::SystemTable`, found struct `uefi::table::SystemTable`
   |
   = note: expected mutable reference `&mut uefi::table::system::SystemTable<uefi::table::system::Boot>`
              found mutable reference `&mut uefi::table::SystemTable<uefi::table::Boot>`
   = note: perhaps two different versions of crate `uefi` are being used?

error[E0599]: no method named `expect_success` found for enum `Result` in the current scope
  --> uefi-test-runner\src\main.rs:26:34
   |
26 |     uefi_services::init(&mut st).expect_success("Failed to initialize utilities");
   |                                  ^^^^^^^^^^^^^^ method not found in `Result<uefi::result::completion::Completion<()>, uefi::result::error::Error>`r>`
   |
  ::: C:\Users\Daniel\Documents\GitHub\uefi-rs\src\result\mod.rs:44:8
   |
44 |     fn expect_success(self, msg: &str) -> Output;
   |        -------------- the method is available for `Result<uefi::result::completion::Completion<()>, uefi::result::error::Error>` here
   |
   = help: items from traits can only be used if the trait is in scope
help: the following trait is implemented but not in scope; perhaps add a `use` for it:
   |
14 | use uefi::result::ResultExt;
   |

Some errors have detailed explanations: E0308, E0599.
For more information about an error, try `rustc --explain E0308`.
error: could not compile `uefi-test-runner` due to 2 previous errors
```

GabrielMajeri, 2021-08-03:
Using the [`cargo tree`](https://doc.rust-lang.org/cargo/commands/cargo-tree.html) reveals that in this scenario you end up with two instances of the `uefi-rs` crate in the dependency tree:

- `uefi-test-runner` depends on `uefi` from Git
- `uefi-test-runner` also depends on `uefi-services` from Git...
  - ... which depends on `uefi` version 0.11 from crates.io

The solution is to use some [Cargo overrides](https://doc.rust-lang.org/cargo/reference/overriding-dependencies.html) instead to make sure you replace `uefi` package with the Git version all along the dependency tree.

CraftyDH, 2021-08-04:
Thanks @GabrielMajeri that worked.

closed 2021-08-04 by CraftyDH