rust-osdev/uefi-rs #287: Unexpected output from `DebugSupport` functions

timrobertsdev, 2021-09-15:
QEMU arch: `x86_64`
QEMU smp: `4`

When testing under QEMU, during the `DebugSupport` tests, the processor ISA is reported as `EBC` instead of the expected `X64` (or variant of). I added support to retrieve the `maximum_processor_index` which reports `0` instead of the expected `3` as well. This behavior occurs with an unmodified `master` as well as my dev branch.

The specification states:  

> It is expected that there will typically be two instances of the EFI Debug Support protocol in the system. 
One associated with the native processor instruction set (IA-32, x64, ARM, RISC-V, or Itanium processor 
family), and one for the EFI virtual machine that implements EFI byte code (EBC).
While multiple instances of the EFI Debug Support protocol are expected, there must never be more than 
one for any given instruction set.

I wrote a test to find all handles that implement the `DebugSupport` protocol using `BootServices::find_handles()` but the only one it found was the `Handle` with the `EBC` version of `DebugSupport`. I tried again by enumerating over every handle, looking for the `DebugSupport` GUID, but only found the same one as before.

Does QEMU's firmware implement the `X64` version? If so, does it need to be activated somehow? My google-fu is failing me, most results end up being this repo, the UEFI Spec, or the EDK2 sources.

timrobertsdev, 2021-09-15:
Did a little more looking, got OVMF compiled in debug mode so that it spits out logs, can confirm that the `DebugSupport` protocol interface is only installed once by the firmware, while it's loading the EBC driver.

timrobertsdev, 2021-09-15:
Hilariously found the following in the `edk2` source:
```
/**
  Returns the maximum value that may be used for the ProcessorIndex parameter in
  RegisterPeriodicCallback() and RegisterExceptionCallback().

  Hard coded to support only 1 processor for now.

  @param  This                  A pointer to the EFI_DEBUG_SUPPORT_PROTOCOL instance.
  @param  MaxProcessorIndex     Pointer to a caller-allocated UINTN in which the maximum supported
                                processor index is returned. Always 0 returned.

  @retval EFI_SUCCESS           Always returned with **MaxProcessorIndex set to 0.

**/
EFI_STATUS
EFIAPI
GetMaximumProcessorIndex (
  IN EFI_DEBUG_SUPPORT_PROTOCOL       *This,
  OUT UINTN                           *MaxProcessorIndex
  )
{
  *MaxProcessorIndex = 0;
  return EFI_SUCCESS;
}
```
So even if OVMF loaded the generic `DebugSupport` driver, we wouldn't be able to get a useful result. Closing this issue as it's not our issue.

closed 2021-09-15 by timrobertsdev

timrobertsdev, 2021-09-15:
3rd comment in as many hours, but I modified OVMF to include the DebugSupportDxe driver, the driver loads correctly, and we can find it correctly while testing. `maximum_processor_index` is still returned as `0`, of course. So, partially solved but still not our issue.

GabrielMajeri, 2021-09-15:
@timrobertsdev it seems the best course of action would be to [report this issue](https://github.com/tianocore/tianocore.github.io/wiki/Reporting-Issues) to the EDKII developer's (in case it's not already tracked), then wait until it gets fixed or submit a patch.