rust-osdev/uefi-rs #289: Document how to debug an application built with uefi-rs

nicholasbishop, 2021-09-15:
Splitting this out from https://github.com/rust-osdev/uefi-rs/issues/285:

> I've been attempting to investigate why, but I've been having a little trouble getting `rust-gdb` and `rust-lldb` to recognize the symbols in the `uefi-test-runner.efi` binary. Oddly enough, `objdump -f`reports that it is flagged as `HAS_DEBUG`, but `objdump -t` wasn't able to find any symbols. Attemping `image dump symtab` in `rust-lldb` results in `num-symbols = 0`. I've probably got something setup incorrectly on my end, any advice would be greatly appreciated.

I don't know the answer to this, but it would be nice to have a straightforward way to debug an application built with uefi-rs running under qemu. I'm not sure if there is something that currently works and it just needs some documentation, or if there's work needed in the toolchain.

timrobertsdev, 2021-09-15:
I found [this article](https://xitan.me/posts/rust-uefi-runtime-driver/), but their approach doesn't seem to work. It finds the base address and address of each section, but still doesn't provide any symbols to gdb. Then I found a `uefi_test_runner_xxxx.pdb` file alongside a `uefi_test_runner_xxxx.efi` file in `target/{arch}/debug/deps/`, but lldb doesn't seem to want to load it, maybe the location of the `.pdb` isn't in the executable? I'm starting to lean towards it being a toolchain issue. I'll try building EDK2 and OVMF using their LLVM/Clang toolchain and see if it outputs usable debug info first.

no92, 2021-10-03:
Previous versions of `rustc` and/or `cargo` worked flawlessly as described in the article above, but this doesn't work with recent version. I don't remember whether the `.pdb` files were always there, but in any case they don't include any symbols now.

IIRC, the spec requires that UEFI apps tick a few boxes, e.g. the subsystem type, but it also states that they ought to be stripped, hence this behavior.

timrobertsdev, 2021-10-03:
The executable should be stripped, but applications built in C with EDK2 in `DEBUG` mode do output `.map` and `.pdb` files for debugging. Source: https://edk2-docs.gitbook.io/edk-ii-build-specification/12_build_changes_and_customizations/121_building_for_debug

With that in mind, it sounds like this might be worth filing a bug for the `x86_64-unknown-uefi` target?

no92, 2021-10-03:
I'm in the process of bisecting compiler versions to determine where the old behavior of having symbol tables was lost.

no92, 2021-10-03:
The last build that provides symbol tables is `nightly-2020-11-14-x86_64-unknown-linux-gnu`, while `nightly-2020-11-15-x86_64-unknown-linux-gnu` doesn't.

I'll try to do some digging as to why that happens, but I don't know the first thing about rustc or cargo code so I don't expect to find the solution TBH.

### Methodology
I took `uefi-test-runner` @ `v0.6.0` as a testbed. Rust nightlies tried ranged from `2020-10-20` to `2021-10-04`. For getting versions of Rust, I used these commands, substituting the dates accordingly:
```
rustup toolchain add nightly-2020-11-14 --profile minimal
rustup default nightly-2020-11-14
rustup component add rust-src
```

For building, `./build.py build` was used. In order to detect whether symbol tables are present, the following commands were run in gdb via a `.gdbinit` file:
```
add-symbol-file ../target/x86_64-unknown-uefi/debug/uefi-test-runner.efi
info functions
```

no92, 2021-10-03:
Bisecting leads to rust-lang/rust#78959, which is also linked to by rust-lang/rust#87157 where the author of the blog post above reports the same issue.

The solution seems to be to do `rustc +nightly -Z unstable-options --print target-spec-json --target x86_64-unknown-uefi | sed -e 's/"is-like-msvc": true/"is-like-msvc": false/g' > x86_64-unknown-uefi-debug.json`.

timrobertsdev, 2021-10-03:
Can confirm the above solution works, with the additional step of removing "is-builtin" from the target file. `build.py` would need to be updated to use the custom target as well. 

I'll dive into debugging something in QEMU tonight, but this looks like it solves the issue completely as gdb can see all symbols and map them back to the source files.

nicholasbishop, 2021-10-04:
I'm not seeing quite the same thing as you with the builtin x86_64-unknown-uefi target. I do get what seems to be a valid pdb file generated under the `deps` directory. I dumped the symbols in it with the `pdb_symbols` utility in the [pdb](https://github.com/willglynn/pdb) crate, and it looks correct. (`rustc 1.57.0-nightly (9dbb26efe 2021-10-03`.)

So if I'm understanding correctly, rustc used to include DWARF-style debug sections in the executable, and now it outputs a pdb. Which does seem like reasonable behavior.

lldb seems like it supports loading pdb files (either with a "native" pdb loader on Windows, or a built-in loader for other platforms), but in testing on Linux I haven't had any luck getting it to successfully load symbols.

timrobertsdev, 2021-10-04:
Yeah, that's the same issue that I was having with the pdb files. I tested on both linux and windows(mingw) and was unable to correctly load any symbols. I'll give MSVC a try. 

With @no92's solution, gdb and lldb in linux are able to recognize the symbols enough to say they have successfully set a breakpoint, but they won't actually break with `br` or `hbr`. Maybe the addresses aren't correct and need to be fixed up before debugging?

no92, 2021-10-04:
@nicholasbishop wrote:
> I do get what seems to be a valid pdb file generated under the deps directory. I dumped the symbols in it with the pdb_symbols utility in the pdb crate, and it looks correct.

I had checked it with some pdb tool I frankly can't remember right now, and I didn't find anything resembling symbols. It was, however, a valid pdb file. As it was some random tool pulled from GitHub, it could just be wrong and the Rust crate right. Oops.

@timrobertsdev wrote:
> With @no92's solution, gdb and lldb in linux are able to recognize the symbols enough to say they have successfully set a breakpoint, but they won't actually break with br or hbr. Maybe the addresses aren't correct and need to be fixed up before debugging?

As EFI apps are relocatable, the base address changes. This could be read out via the LoadedImage Protocol, but transferring that info from the remote to the debugger host would have to take place somehow. Alternatively, the instructions in the [article](https://xitan.me/posts/rust-uefi-runtime-driver/) work for me, I'm not sure why it's segfaulting for you. I use the python script via a `.gdbinit` in gdb itself as well as some VSCode interface that ultimately uses gdb, too.

timrobertsdev, 2021-10-04:
@no92 Nevermind, I got it working. The mistake I was making was attempting to run the `dbg` gdbscript while QEMU itself was halted and waiting for debugger, instead of the `wait_for_debugger()` fn outlined in that article.

vitamiin, 2022-09-29:
I believe I've taken all the steps mentioned above yet the problem persists -- GDB is not able to recognize any symbols (which I doubt exist). I might've missed something. Any help would be much appreciated.

**What I did:**
- Created a custom target, and in its .json file, I removed the is_builtin line, as well as set is-like-msvc to true.
- Rolled back the nightly version to `nightly-2020-11-14` (used rustup default and updated `rust-toolchain.toml`)
- Built the project using the custom target (which is specified in my `.cargo/config` file and was indeed recognized)

no92, 2022-09-29:
While I haven't used this in a while, rolling back to an old nightly version should not be necessary. Using the script provided in the blog post linked above is necessary IIRC.

vitamiin, 2022-09-29:
Unfortunately, it still doesn't work for me. We're talking about `load_symbols.py`, right?

vitamiin, 2022-10-08:
I have successfully solved the problem by adding the following build (under the [build] section) flag in .cargo/config:

    "-C", "link-args=/debug:dwarf"

Thanks to the following video:
https://www.youtube.com/watch?v=2YAgDJTs9So

This might be trivial to some however I wasn't aware of this link argument in the first place.
All symbols are now recognized by gdb.

DianaNites, 2022-11-20:
How did you get it working with gdb? I can't seem to figure it out

phip1611, 2023-07-20:
> ```
> "-C", "link-args=/debug:dwarf"
> ```

This help that `objdump` displays debug symbols. GDB however still doesn't find them.

GabrielMajeri, 2023-08-07:
I've been digging around this issue recently, here's what I've discovered so far. I've been trying to use LLDB, since it is the only debugger which can read PDB files (in theory) and connect to QEMU using GDB's remote debug protocol.

I've tried to follow along with the instructions from [this article](https://wiki.osdev.org/Debugging_UEFI_applications_with_GDB) on the osdev.org wiki. I've modified the `xtask/src/qemu.rs` file to pass the `-s` flag to QEMU (i.e. I've added a line with `cmd.arg("-s");`) in order to activate the GDB debug port, then I've modified the `uefi-test-runner/src/main.rs` file to display the address at which the binary is loaded:

```rust
{
    let loaded_image = st.boot_services()
        .open_protocol_exclusive::<LoadedImage>(image)
        .expect("Failed to open LoadedImage protocol");
    println!("Image base: {:#X}", loaded_image.info().0 as usize);
}

let wait = true;
while wait { }
```

When running the app with `cargo xtask run`, it displays a message similar to `Image base: 0xDD89000`.

Now, I ran the following commands in LLDB, from the `target/x86_64-unknown-uefi/debug/deps` directory:
- `target create uefi_test_runner-e4bed4b0ac1a4cab.efi` &mdash; loads the EFI binary and sets it as the default debug target.
- `target symbols add uefi_test_runner-e4bed4b0ac1a4cab.pdb` &mdash; loads the debug symbols from the PDB file.
- `gdb-remote localhost:1234` &mdash; connects to the running QEMU instance (**Note**: sometimes I had to run it twice to get it to work).
- `process continue` &mdash; allow the code to run until it gets into the infinite loop we've set up in the main function.
- `process interrupt` &mdash; pause inside the `while` loop inside the main function.
- `image load --file uefi_test_runner-e4bed4b0ac1a4cab.efi .text 0xDD89000` &mdash; tell LLDB at which address the EFI binary's `.text` section starts.

At this point, everything should be set up for debugging. Unfortunately, LLDB just crashes:
```
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Program arguments: lldb
 #0 0x00007f9ba8e3fd01 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0xe3fd01)
 #1 0x00007f9ba8e3da3e llvm::sys::RunSignalHandlers() (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0xe3da3e)
 #2 0x00007f9ba8e40236 (/lib/x86_64-linux-gnu/libLLVM-14.so.1+0xe40236)
 #3 0x00007f9ba7842520 (/lib/x86_64-linux-gnu/libc.so.6+0x42520)
 #4 0x00007f9bb2ca5f75 (/lib/x86_64-linux-gnu/liblldb-14.so.1+0xaa5f75)
 #5 0x00007f9bb2caa78e (/lib/x86_64-linux-gnu/liblldb-14.so.1+0xaaa78e)
 #6 0x00007f9bb2cab262 (/lib/x86_64-linux-gnu/liblldb-14.so.1+0xaab262)
 #7 0x00007f9bb2ca6c91 (/lib/x86_64-linux-gnu/liblldb-14.so.1+0xaa6c91)
 #8 0x00007f9bb2ca6d32 (/lib/x86_64-linux-gnu/liblldb-14.so.1+0xaa6d32)
 #9 0x00007f9bb2caa613 (/lib/x86_64-linux-gnu/liblldb-14.so.1+0xaaa613)
#10 0x00007f9bb2cab2ef (/lib/x86_64-linux-gnu/liblldb-14.so.1+0xaab2ef)
#11 0x00007f9bb2ca6c91 (/lib/x86_64-linux-gnu/liblldb-14.so.1+0xaa6c91)
#12 0x00007f9bb2cab7a0 (/lib/x86_64-linux-gnu/liblldb-14.so.1+0xaab7a0)
#13 0x00007f9bb2cab3d6 (/lib/x86_64-linux-gnu/liblldb-14.so.1+0xaab3d6)
#14 0x00007f9bb2ca6c91 (/lib/x86_64-linux-gnu/liblldb-14.so.1+0xaa6c91)
#15 0x00007f9bb2c88d16 (/lib/x86_64-linux-gnu/liblldb-14.so.1+0xa88d16)
#16 0x00007f9bb2c8936d (/lib/x86_64-linux-gnu/liblldb-14.so.1+0xa8936d)
#17 0x00007f9bb2c8587b (/lib/x86_64-linux-gnu/liblldb-14.so.1+0xa8587b)
#18 0x00007f9bb2c85263 (/lib/x86_64-linux-gnu/liblldb-14.so.1+0xa85263)
#19 0x00007f9bb2c8b078 (/lib/x86_64-linux-gnu/liblldb-14.so.1+0xa8b078)
#20 0x00007f9bb260e6ec (/lib/x86_64-linux-gnu/liblldb-14.so.1+0x40e6ec)
#21 0x00007f9bb25de1ba (/lib/x86_64-linux-gnu/liblldb-14.so.1+0x3de1ba)
#22 0x00007f9bb25de004 (/lib/x86_64-linux-gnu/liblldb-14.so.1+0x3de004)
#23 0x00007f9bb27f5f1e (/lib/x86_64-linux-gnu/liblldb-14.so.1+0x5f5f1e)
#24 0x00007f9bb27f5b90 (/lib/x86_64-linux-gnu/liblldb-14.so.1+0x5f5b90)
#25 0x00007f9bb27e8d82 (/lib/x86_64-linux-gnu/liblldb-14.so.1+0x5e8d82)
#26 0x00007f9bb27ea36f (/lib/x86_64-linux-gnu/liblldb-14.so.1+0x5ea36f)
#27 0x00007f9bb2781b2f (/lib/x86_64-linux-gnu/liblldb-14.so.1+0x581b2f)
#28 0x00007f9bb2783c29 (/lib/x86_64-linux-gnu/liblldb-14.so.1+0x583c29)
#29 0x00007f9bb27ba671 (/lib/x86_64-linux-gnu/liblldb-14.so.1+0x5ba671)
#30 0x00007f9bb274184c (/lib/x86_64-linux-gnu/liblldb-14.so.1+0x54184c)
#31 0x00007f9bb25eb03f (/lib/x86_64-linux-gnu/liblldb-14.so.1+0x3eb03f)
#32 0x00007f9bb26c1912 (/lib/x86_64-linux-gnu/liblldb-14.so.1+0x4c1912)
#33 0x00007f9bb260a83f (/lib/x86_64-linux-gnu/liblldb-14.so.1+0x40a83f)
#34 0x00007f9bb25eb73c (/lib/x86_64-linux-gnu/liblldb-14.so.1+0x3eb73c)
#35 0x00007f9bb26c33d9 (/lib/x86_64-linux-gnu/liblldb-14.so.1+0x4c33d9)
#36 0x00007f9bb23f5eda lldb::SBDebugger::RunCommandInterpreter(bool, bool) (/lib/x86_64-linux-gnu/liblldb-14.so.1+0x1f5eda)
#37 0x0000000000407d4a (/usr/lib/llvm-14/bin/lldb+0x407d4a)
#38 0x0000000000408f85 (/usr/lib/llvm-14/bin/lldb+0x408f85)
#39 0x00007f9ba7829d90 __libc_start_call_main ./csu/../sysdeps/nptl/libc_start_call_main.h:58:16
#40 0x00007f9ba7829e40 call_init ./csu/../csu/libc-start.c:128:20
#41 0x00007f9ba7829e40 __libc_start_main ./csu/../csu/libc-start.c:379:5
#42 0x0000000000404125 (/usr/lib/llvm-14/bin/lldb+0x404125)
Segmentation fault (core dumped)
```

I'm guessing it tries to parse the PDB file and the current stack frame, but it triggers some bug somewhere.

I haven't submitted a bug report to LLDB yet, I'm going to play around with this setup for a bit longer.

stevefan1999-personal, 2023-08-25:
@GabrielMajeri I managed to replicate your process in VSCode and LLDB shell as well

lf-, 2023-09-17:
The pdb files not being copied is a cargo bug: https://github.com/rust-lang/cargo/pull/5179 did not include -uefi as a target to copy pdbs for. I don't feel like filing an actual bug right now; I'm presently testing just patching it and I may file a PR later.

FuuuOverclocking, 2024-01-17:
After a week of struggle, I finally realized that using LLDB to debug UEFI applications is far from mature. And I still cannot let the rustc and lld to output dwarf debug symbols. The line below seems to have no effect on me.

```
"-C", "link-args=/debug:dwarf"
```

stemnic, 2024-02-08:
The following addition in `.cargo/config` worked for me. The resulting .efi file now has dwarf symbols which can be used with gdb.
```toml
[target.x86_64-unknown-uefi]
rustflags = ["-C", "link-args=/debug:dwarf"]
```

tandasat, 2025-02-23:
Unfortunately, this workaround (https://github.com/rust-osdev/uefi-rs/issues/289#issuecomment-1934079770) is broken since this commit:
https://github.com/rust-lang/rust/commit/562a85579e279e7827eb9785055322e0942b6371#diff-aa810a3be0834da891b171a8b04b09d0d3bbb76ac27c757cd232235099e062d2R2923-R2926

Build fails with this error message:
```
...
error: failed to run `rustc` to learn about target-specific information

Caused by:
  process didn't exit successfully: ... (exit code: 1)
  --- stderr
  error: Error loading target specification: `linker_flavor` must be `msvc` if and only if `is_like_msvc` is set. Run `rustc --print target-list` for a list of built-in targets
```

sudoBash418, 2025-03-07:
I've been able to get GDB + QEMU working with the following method (using `rustc 1.87.0-nightly (f9e0239a7 2025-03-04)`):

Step 1: create a custom target JSON:
```
# dump original target json to file
rustc +nightly -Z unstable-options --print target-spec-json --target x86_64-unknown-uefi > x86_64-unknown-uefi-debug.json
# switch debuginfo-kind from pdb to dwarf
sed -i 's/"debuginfo-kind": "pdb"/"debuginfo-kind": "dwarf"/' x86_64-unknown-uefi-debug.json
# switch split-debuginfo from packed to unpacked
sed -i 's/"packed"/"unpacked"/' x86_64-unknown-uefi-debug.json
```

Step 2: add the following to `.cargo/config.toml`:
```
[build]
rustflags = ["-C", "link-args=/debug:dwarf"]
```

Step 3: build with the following command (set `build-std` options to whatever you need):
```
cargo +nightly -Z build-std=core,alloc build --target ./x86_64-unknown-uefi-debug.json
```

Step 4: run QEMU using `-gdb tcp::1337 -S`

Step 5: run `llvm-readobj` to get section offsets
```shell
$ llvm-readobj --sections target/x86_64-unknown-uefi-debug/debug/myprogram.efi | rg 'Name|VirtualAddress'
    Name: .text (2E 74 65 78 74 00 00 00)
    VirtualAddress: 0x1000
    Name: .rdata (2E 72 64 61 74 61 00 00)
    VirtualAddress: 0x49000
    Name: .data (2E 64 61 74 61 00 00 00)
    VirtualAddress: 0x57000
[...]
```

Step 6: run GDB using `rust-gdb` (my image base was `0xdfa8000`)
```
(gdb) add-symbol-file target/x86_64-unknown-uefi-debug/debug/myprogram.efi -s .text 0xdfa8000+0x1000 -s .rdata 0xdfa8000+0x49000 -s .data 0xdfa8000+0x57000
add symbol table from file "target/x86_64-unknown-uefi-debug/debug/myprogram.efi" at
	.text_addr = 0xdfa9000
	.rdata_addr = 0xdff1000
	.data_addr = 0xdfff000
(y or n) y
Reading symbols from target/x86_64-unknown-uefi-debug/debug/myprogram.efi...
(gdb) target remote localhost:1337
Remote debugging using localhost:1337
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x000000000000fff0 in ?? ()
(gdb) hbreak efi_main
Hardware assisted breakpoint 1 at 0xdfaa84c: file crates/myprogram/src/main.rs, line 103.
(gdb) c
Continuing.

Breakpoint 1, myprogram::main (internal_image_handle=..., internal_system_table=0xf9ec018) at crates/myprogram/src/main.rs:103
103	#[entry]
(gdb)
```

At this point I'm able to step through code and print local variables, etc. (in all crates, not just my own code)

(note: if using pwndbg, you might want to run `set osabi none` and `set kernel-vmmap monitor` in gdb as well)

Aside: it looks like the step of determining the image base can probably be automated via a debugger script: https://uefi.org/specs/UEFI/2.10/18_Protocols_Debugger_Support.html#efi-debug-support-table

---

Unfortunately LLDB seems to hang at "Manually indexing DWARF" when I run it via `rust-lldb -f target/...`.
I got LLDB running with the "packed" debuginfo option, but then it had incomplete debug info and complained about missing DWO files :/

tmvkrpxl0, 2025-05-22:
@sudoBash418 I wrote a script for automatically figuring it out. tho it's very imperfect and probably only work for my project for now.
The problem is that there are lots of image handles. Since my uefi binary includes binary files, it was biggest one and I picked mine based on size. But it should be adjusted for other's use. During test I observed my image is last one. I'm not sure if it's always the last one but it's good guess.
Here is the script:
```py
from subprocess import check_output

import gdb

# Constants
MAGIC = 0x5453595320494249  # "IBI SYST"
ALIGNMENT = 0x400000  # 4 MiB
START = 0x0
END = 0x100000000  # 4 GiB
TARGET_NAME = "x86_64-unknown-uefi-debug"
BINARY_NAME = "uefi-still-alive.efi"
DEBUG_IMAGE_INFO_GUID = (0x49152E77,0x1ADA,0x4764,
                         0xB7,0xA2,0x7A,0xFE,0xFE,0xD9,0x5E,0x8B )

def read_u64(addr):
    return int(gdb.execute(f"x/gx {addr:#x}", to_string=True).split(":")[1], 16)

def read_u32(addr):
    return int(gdb.execute(f"x/wx {addr:#x}", to_string=True).split(":")[1], 16)

def read_u16(addr):
    return int(gdb.execute(f"x/hx {addr:#x}", to_string=True).split(":")[1], 16)

def read_u8(addr):
    return int(gdb.execute(f"x/bx {addr:#x}", to_string=True).split(":")[1], 16)

def find_system_table():
    for addr in range(START, END, ALIGNMENT):
        try:
            val = read_u64(addr)
            if val == MAGIC:
                system_table_ptr = read_u64(addr + 8)
                print(f"[+] Found EFI signature at 0x{addr:016X}")
                print(f"[+] EFI System Table is at 0x{system_table_ptr:016X}")
                return system_table_ptr
        except Exception as e:
            print(f"[!] Failed to read at 0x{addr:016X}: {e}")
    return None

def find_base():
    system_table = find_system_table()
    config_table_entry_count = read_u64(system_table + 104)
    config_table_entries = read_u64(system_table + 112)
    print(f"[+] Config entries: {config_table_entry_count}")
    debug_info_header_address = -1
    for i in range(0, config_table_entry_count):
        entry_address = config_table_entries + i * 24
        guid = read_guid(entry_address)
        print(f"[+] Checking entry with guid {guid}")
        if guid != DEBUG_IMAGE_INFO_GUID:
            continue
        print(f"[+] Found debug info header at 0x{entry_address:016X}")
        debug_info_header_address = read_u64(entry_address + 16)
        break

    if debug_info_header_address == -1:
        print("[!] Failed to find debug info header")
        return None

    debug_info_table_size = read_u32(debug_info_header_address + 4)
    debug_info_table_address = read_u64(debug_info_header_address + 8)

    normal_infos = []
    for i in range(0, debug_info_table_size):
        current_debug_info_pointer_address = debug_info_table_address + i * 8
        current_debug_info_pointer = read_u64(current_debug_info_pointer_address)
        debug_info_type = read_u32(current_debug_info_pointer)
        if debug_info_type == 1:
            protocol_address = read_u64(current_debug_info_pointer + 8)
            handle_address = read_u64(current_debug_info_pointer + 16)
            normal_infos.append((protocol_address, handle_address))

    print(f"[+] Found {debug_info_table_size} entries at {debug_info_table_address:X}")

    image_infos = []
    for (protocol_address, handle_address) in normal_infos:
        image_base = read_u64(protocol_address + 64)
        image_size = read_u64(protocol_address + 72)
        image_infos.append((image_base, image_size))

    biggest = max(image_infos, key=lambda x: x[1])

    return biggest

def read_guid(address):
    time_low = read_u32(address)
    time_mid = read_u16(address + 4)
    time_high_and_version = read_u16(address + 6)
    clock_seq_high_and_reserved = read_u8(address + 8)
    clock_seq_low = read_u8(address + 9)
    node1 = read_u8(address + 10)
    node2 = read_u8(address + 11)
    node3 = read_u8(address + 12)
    node4 = read_u8(address + 13)
    node5 = read_u8(address + 14)
    node6 = read_u8(address + 15)

    return time_low, time_mid, time_high_and_version, clock_seq_high_and_reserved, clock_seq_low, node1, node2, node3, node4, node5, node6

def find_sections():
    output = check_output(
        f"llvm-readobj-18 --sections target/{TARGET_NAME}/debug/{BINARY_NAME} | rg 'Name|VirtualAddress'",
        shell=True
    ).decode("utf-8")
    lines = output.split("\n")
    text_line = lines[1]
    rdata_line = lines[3]
    data_line = lines[5]

    text_number = text_line[20:]
    rdata_number = rdata_line[20:]
    data_number = data_line[20:]

    return text_number, rdata_number, data_number

def load_symbols():
    base, size = find_base()
    print(f"[+] Found base 0x{base:X}")
    text, rdata, data = find_sections()
    print(f"[+] Found sections:")
    print(f"[+]   {text}")
    print(f"[+]   {rdata}")
    print(f"[+]   {data}")
    gdb.execute(f"add-symbol-file target/{TARGET_NAME}/debug/{BINARY_NAME} -s .text 0x{base}+{text} -s .rdata 0x{base}+{rdata} -s .data 0x{base}+{data}")

gdb.execute("target remote :1337")
print("[*] Loading symbols...")
load_symbols()
```

The way it works is that it runs `llvm-readobj-18` to fetch text, rdata and data offsets. and then it consults qemu tot get image base. and then add those 2 together to load image as you specified.
One weird thing I've discovered is that it may crash qemu for some reason. it works fine on my desktop but not on my laptop.