rust-osdev/uefi-rs #307: newtype_enum! macro should have it's own crate

necauqua, 2021-11-04:
It is actually an incredibly useful macro for autogenerating 'enums' and their debug for various codes that you receive from outside of Rust.
It's similar in its usefulness to the `bitflags!` macro, and it's a shame I have to copy it to my code from here instead of using a crate.
Although I didn't really search if anyone made it (or something similar) a separate crate already

GabrielMajeri, 2021-12-11:
> It's similar in its usefulness to the bitflags! macro, and it's a shame I have to copy it to my code from here instead of using a crate.

Agreed! Didn't realize it could be so useful outside the project as well.

As far as I know, it shouldn't be any problem for it to be published as a separate crate, as long as this is done under the same open source license. If anyone is interested in doing so, feel free to continue the discussion here.

> Although I didn't really search if anyone made it (or something similar) a separate crate already

I'm not sure about this either, a double-check would be indeed useful to avoid creating something which already exists.

nicholasbishop, 2021-12-12:
This crate seems like a similar idea: https://docs.rs/fake-enum/latest/fake_enum/

nicholasbishop, 2023-03-04:
Here's another crate with a similar feature: https://docs.rs/open-enum/latest/open_enum/

Given that there are a few crates that do similar things, I'll go ahead and close this issue.

closed 2023-03-04 by nicholasbishop