rust-osdev/uefi-rs #339: Many, many compilation issues (especially with crates.io versions)

amyipdev, 2022-01-14:
The only solution that works is cloning the repo and using it directly as set up. Attempting to make a standalone project causes a need for `-Z build-std` to be run with cargo; using crates.io versions causes the `uefi` crate to be extremely messed up and `uefi-services` to be unable to compile (it needs core::arch::asm, not just asm, apparently). Running nightly-x86_64-unknown-linux-gnu as my base toolchain.

amyipdev, 2022-01-14:
I've found a configuration that makes it work: repository-based uefi-services, crates.io-based uefi.

nicholasbishop, 2022-01-15:
Thanks for the heads up! I've released new versions of the uefi and uefi-services crates (https://github.com/rust-osdev/uefi-rs/issues/340), so your issue should be fixed.

I also filed https://github.com/rust-osdev/uefi-rs/issues/343 to add some periodic CI for the released versions of crates (currently we just test master) to hopefully prevent issues like this from lingering too long in the future.

closed 2022-01-15 by nicholasbishop