rust-osdev/uefi-rs #343: Consider periodically running CI on the current released version of the packages

nicholasbishop, 2022-01-15:
Since uefi-rs is used with the nightly compiler, sometimes the released versions of the packages suddenly stop compiling when changes to the compiler hit the nightly channel. See https://github.com/rust-osdev/uefi-rs/issues/339 for an example.

To some extent we should already know when this happens, since usually the `master` branch will also break, but it would still be nice to have a reminder in the form of periodic "CI failed" emails.

GabrielMajeri, 2022-01-15:
Since https://github.com/rust-osdev/uefi-rs/pull/337 will make the template project use the published versions of the crate, and since we build the template in CI, maybe this will be enough?

nicholasbishop, 2022-01-15:
I think that would not work because of the [`patch.crates-io`](https://github.com/rust-osdev/uefi-rs/blob/master/Cargo.toml#L48) section in the root crate. That overrides crates.io deps to use the local version instead. Which I think is not a bad thing, since otherwise we could accidentally break compilation of the template against master, just not sufficient to cover this use case.

closed 2022-01-24 by GabrielMajeri