rust-osdev/uefi-rs #346: Consider renaming master branch to main

nicholasbishop, 2022-01-15:
There's a [general trend](https://sfconservancy.org/news/2020/jun/23/gitbranchname/) to rename master branches to main, and github now makes this quite easy to do in the settings: https://github.com/github/renaming#rename-existing (takes care of things like pointing open PRs to the new branch name).

wbx, 2022-01-16:
personally, I don't agree with the renaming trend, it's important to deal with issues regarding racism (which this was mostly about), but this has nothing to do with anything racist within the context, nor do I think anyone was seriously offended by it, and the trend serves as nothing but a distraction to the real issues, and creates unnecessary work for developers.

I myself use "trunk", not for any particular reason, just personal preference.

GabrielMajeri, 2022-01-19:
@wbx I've noted your objections, but I don't have an issue with this rename.

> creates unnecessary work for developers

Well, considering it's an open source/volunteering-based project, it's up to each one of us to decide what we'd like to dedicate our time on. It didn't take more than 3-4 minutes.

> EDIT: I don't see how this is off-topic? I'm merely voicing my opinion on the matter.

No idea either. I'm not familiar with GitHub's moderation policies 🤷‍♂️

GabrielMajeri, 2022-01-19:
Closing this issue since the `master` branch has now been renamed to `main`.

closed 2022-01-19 by GabrielMajeri

nicholasbishop, 2022-01-19:
Re the moderation, that was me.

My apologies, on rereading your message I was wrong to mark it as offtopic. My initial reaction was to note that you hadn't previously been involved in this project (or any other public project on github), and I immediately assumed that you were just searching github projects for master->main tickets to comment on, as people have unfortunately done that on other projects. That was unfair of me to assume, so again I apologize and I've unhidden your comment.

You may not have had the additional context that the person filing this issue (me) is a co-maintainer of this project, so I'm not just going around trying to create extra work for developers. :)

wbx, 2022-01-19:
> Re the moderation, that was me.
>
> My apologies, on rereading your message I was wrong to mark it as offtopic. My initial reaction was to note that you hadn't previously been involved in this project [...]

no worries!
can assure you my intention wasn't going around github to random projects to comment on master -> main issues, though I cannot blame you for thinking so... I merely happened to come across the issue and decided to voice my opinion.
Though looking back I'm just being pedantic, the issue has already been debated a lot in other places and it really doesn't matter, especially not in this case where it doesn't need a lot of work to rename the branch as in this project.