rust-osdev/uefi-rs #372: implement `locate_handle_buffer` function?

AtsukiTak, 2022-02-26:
Hi there!

It seems that `BootServices` does not implement `locate_handle_buffer` function. Is there any reason to not do it?

I just tried and found that making a tidy API is not easy. (My API is like below)

```rust
pub fn locate_handle_buffer(&self, search_ty: SearchType) -> Result<(*mut Handle, usize)>
```

But it seems to work correctly. So if it is the only reason to not do, I would like to discuss API :) Thanks!

nicholasbishop, 2022-02-26:
> It seems that BootServices does not implement locate_handle_buffer function. Is there any reason to not do it?

No reason, just hasn't been done yet. Thanks for looking into it :)

I think a good model to look at for implementing this is [`protocols_per_handle`](https://github.com/rust-osdev/uefi-rs/blob/48a7261d7890bc11d9a33dcb0521276f3199a6d6/src/table/boot.rs#L760). That shows an example of wrapping the pointer into a slice, and also ensuring that the memory is freed on drop.

AtsukiTak, 2022-02-27:
Before I will try to implement `locate_handle_buffer`, I would like to fix https://github.com/rust-osdev/uefi-rs/pull/374

nicholasbishop, 2022-02-27:
> The memory allocated by the locate_handle_buffer function is not a Handle pointer itself. It is a buffer that contains Handle pointers. So we can free that buffer safely on drop. Is it right?

That's correct, it should be safe to free the buffer containing `Handle` pointers.

closed 2022-03-09 by AtsukiTak