rust-osdev/uefi-rs #384: Release uefi-0.15.0, uefi-services-0.12.0, uefi-macros-0.6.0

nicholasbishop, 2022-03-06:
It's been a couple months since the last release and we've accumulated a fair number of changes (including some correctness fixes), so it's probably time for another release.

GabrielMajeri, 2022-03-07:
Might want to hang on and wait until the `Completion` changes are implemented, or would you rather want us to release these new versions now and save the `Completion` removal for the next bump?

nicholasbishop, 2022-03-09:
No strong opinion either way, happy to wait until the Completion change is in.

sven-eliasen, 2022-03-09:
Hi,

I would love having merged of my PR  #386  for this release.
I dont know when you think you're gonna merge the Completion change but I'll do my best to wrap this up as fast as possible.

GabrielMajeri, 2022-03-13:
I'd say we're pretty ready for a new release, especially since both #361 and #386 got merged. Anything else before the bump?

nicholasbishop, 2022-03-13:
Nothing else that I'm aware of.