rust-osdev/uefi-rs #403: Locale device tree address

Luchangcheng2333, 2022-04-05:
I wonder if there is a way to get the address of the device tree by UEFI in aarch64? It's wonderful if the crate can provide an interface to get the address of device tree.

GabrielMajeri, 2022-04-06:
Hello! You might want to look into navigating the EFI Configuration Table structure, accessible with [the `.config_table()` method](https://docs.rs/uefi/latest/uefi/table/struct.SystemTable.html#method.config_table) on the `SystemTable`. It provides access to a list of descriptors of well-known, predefined "configuration tables", and [according to the UEFI spec](https://uefi.org/sites/default/files/resources/UEFI_Spec_2_9_2021_03_18.pdf) (page 104), you should be able to find a config table entry for Device Tree:

```C
//
// Devicetree table, in Flattened Devicetree Blob (DTB) format
//
#define EFI_DTB_TABLE_GUID \
{0xb1b621d5, 0xf19c, 0x41a5, \
{0x83, 0x0b, 0xd9, 0x15, 0x2c, 0x69, 0xaa, 0xe0}}
```

We have some constant GUIDs [defined](https://github.com/rust-osdev/uefi-rs/blob/3e10cba3505ff582e4c84bb119bda3e97aa12c4b/src/table/cfg.rs#L74) for common tables, but not for DeviceTree (yet). If you're interested in making it easier to access this information for users of the `uefi-rs` crate, we'd welcome a PR adding it to the list 🙂

phip1611, 2025-04-09:
Closing due to inactivity and unclear scope of specific steps

closed 2025-04-09 by phip1611