rust-osdev/uefi-rs #431: Add support for the Simple Network Protocol

GabrielMajeri, 2022-05-16:
This issue tracks the crate's support for the UEFI Simple Network Protocol.

d-sonuga, 2022-10-25:
Hello. I'm new here and I would like to add the SNP Protocol. Do I just get started immediately or are there some things I have to do first?

nicholasbishop, 2022-10-26:
Hi, and welcome :) You can feel free to jump right in. There is a prior PR (linked above) that you can use as a base if you want, but feel free to proceed however your like. Please also feel free to ask any questions you might have.

d-sonuga, 2022-10-26:
Thanks. I'll just get started then.

d-sonuga, 2022-10-30:
Hi. I made quite some progress with the basic wrapper for the SNP but I'm really stumped on the test right now. The receive buffers just never receive. Any suggestions for dealing with this kind of thing?

d-sonuga, 2022-10-31:
I've posted a draft PR here: [https://github.com/rust-osdev/uefi-rs/pull/541](url)

PR #606 (2022-12-16): Implement SIMPLE_NETWORK_PROTOCOL
Fixes #431.

The code is written on top of @d-sonuga's PR, with the main change being in the tests, where an UDP packet is sent to the UDP echo server and the answer is read back.
closed 2022-12-16 by nicholasbishop