rust-osdev/uefi-rs #441: exts feature uses builtin allocator?

hyperswine, 2022-06-12:
I tried building without the `alloc` feature but it still seems to conflict with my own global allocator. Is this intended?

nicholasbishop, 2022-06-12:
That is not intended, you should be able to enable `exts` without `alloc` and use your own allocator.

My initial guess is that maybe you are depending on both the `uefi` and `uefi-services` crates? The `uefi-services` crate depends on `uefi` with the `alloc` feature, so if you want to use a custom allocator you can't use `uefi-services`.

hyperswine, 2022-06-12:
> That is not intended, you should be able to enable `exts` without `alloc` and use your own allocator.
>
> My initial guess is that maybe you are depending on both the `uefi` and `uefi-services` crates? [...]

I unlinked uefi-services and it stopped using its internal allocator. Seems like `use uefi_services` auto links it which makes sense. Thanks

closed 2022-06-12 by hyperswine