rust-osdev/uefi-rs #459: Add mdBook documentation

nicholasbishop, 2022-07-04:
I've started working on some new documentation for uefi-rs, rendered with [mdBook](https://rust-lang.github.io/mdBook). 

Here's the rendered WIP book showing what I have so far:
<https://nicholasbishop.github.io/uefi-rs>

This is intended to complement the crate documentation on docs.rs, helping to fill in some of the gaps between that documentation and the UEFI Specification. Here's roughly what I hope to cover for starters:
1. A simple tutorial to walk through the steps needed to create, build, and run a "hello world" application.
2. Walkthroughs for things like opening protocols, reading files, and displaying graphics.
3. Explanations of important UEFI concepts. The specification is of course the canonical source of information, but it can make for pretty dense reading so I think we can compliment it with some high-level overviews.

Related to this I want to add more standalone examples to the test runner. These can then be included by `mdBook`, which has special syntax for showing parts of a file so that individual parts can be explained.

I'm not quite ready to put up a PR yet, and it might end up being split over a few PRs anyway so figure'd I'd go ahead and make an issue to track the work.

Feedback welcome!

GabrielMajeri, 2022-07-04:
Looks great! Do you want to eventually include it in the main repo, or spin it off as a separate documentation/guide?

nicholasbishop, 2022-07-05:
I can see some pros and cons of both options, but I'm leaning towards the main repo because it's easier to keep stuff in sync -- e.g. if an API that one of the examples use is changed in the `uefi` crate, you'll get a compilation error right away in the example and see that it needs fixing.

closed 2022-09-28 by nicholasbishop