rust-osdev/uefi-rs #468: uefi-rs template not linking on windows

cleanbaja, 2022-07-20:
I tried building a sample UEFI aplication (based on the template), and cargo keeps on gettting stuck on the linking stage. I followed the instructions on [BUILDING.md](https://github.com/rust-osdev/uefi-rs/blob/main/BUILDING.md), and even tried this on another windows machine, but no dice.

GabrielMajeri, 2022-07-22:
Hello. Does the linker or compiler give you any error output which might suggest as to what is going wrong?

Note that we've only developed and tested this crate until now using Linux or WSL - building UEFI apps using `uefi-rs` on Windows isn't really supported, but I'll do my best to remove blocking issues.

cleanbaja, 2022-07-23:
@GabrielMajeri first off, thanks for responding, I really appreciate it. As for the command itself, cargo gives no output, and is stuck building/linking the final UEFI binary. Aside from standard progress indicators, cargo gives no output at all. As a side note, I do have `ld.lld` installed, along with clang and the llvm tools.

GabrielMajeri, 2022-07-26:
Hmm, that's weird. Do you at least see some linker processes starting and running in the background? Is your CPU being used, or is the linker stuck without actually doing anything?

The `cargo xtask build` command will eventually run something like `cargo +nightly build --target x86_64-unknown-uefi -Zbuild-std=core,compiler_builtins,alloc -Zbuild-std-features=compiler-builtins-mem --package uefi --package uefi_app --package uefi-macros --package uefi-services --package uefi-test-runner --features alloc,exts,logger --verbose` (without the `--verbose` part, I've added it now). Could you run it manually and see if you get some more log output?

nicholasbishop, 2022-11-04:
Closing since there haven't been any updates in a while. If you are still running into the issue please feel free to reopen.

closed 2022-11-04 by nicholasbishop

cleanbaja, 2022-11-05:
@nicholasbishop I still didn't find out the cause of the issue, nor have I tested later versions of this library (since I decided to skip rust altogether and use C), so it's fine by me that you closed it